devbis/z03mmc #1: Not (yet) supported in zigbee2mqtt

Ctrl-F4, 2023-10-08:
Great project!  Just successfully flashed this firmware and paired it with zigbee2mqtt. Would be nice if it will be supported by zigbee2mqtt in the future. Is there already an external converter available?

[image]

devbis, 2023-10-08:
Yes, it is not in the repo, but I'll add it soon. 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['LYWSD03MMC'],
    model: 'LYWSD03MMC',
    vendor: 'Xiaomi',
    description: 'Temperature & humidity sensor',
    fromZigbee: [fz.temperature, fz.humidity, fz.battery],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
    exposes: [e.temperature(), e.humidity(), e.battery()],
  
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Ctrl-F4, 2023-10-08:
Thank you for the converter, it is now recognized in zigbee2mqtt although humidity value seems not (yet) reported? 

[image]

I also checked with the z03mmc_heiman firmware and is also not reporting the humidity value?

[image]

devbis, 2023-10-08:
Both values are reported. Check temp and humid clusters are bound in zigbee2mqtt tab. And rebind if checkboxes are different

[image]

Ctrl-F4, 2023-10-08:
Checkboxes are same as yours. Tried to rebind, rejoin, reflash but no humidity readings. Downloaded your precompiled fw and no change . Tried to sniff the data with wireshark but found no data for humidity.     
Don't know how to debug further.

[image]

devbis, 2023-10-09:
Can you read values out of the cluster with dev console in zigbee2mqtt for a device?
[image]

devbis, 2023-10-09:
Please, check firmware from this run: https://github.com/devbis/z03mmc/actions/runs/6455005642
It has increased number of items for reporting and binding and it could help.

Ctrl-F4, 2023-10-09:
> Can you read values from the cluster with dev console in zigbee2mqtt for a device? [image]

Yes this works! But value not updated automatic.  Not yet tried the test fw (6455005642)

Ctrl-F4, 2023-10-09:
> Please, check firmware from this run: https://github.com/devbis/z03mmc/actions/runs/6455005642 It has increased number of items for reporting and binding and it could help.

Same behaviour for this fw ( uploaded z03mmc.bin). Via dev console forcing humidity readout, it shows same value as lcd. Then increase temp and humidity at the same time only temp reading is updated.

devbis, 2023-10-09:
What's on the reporting page? Is humidity configured? Does remove and and set back change anything?

devbis, 2023-10-09:
I have the same parameters and it seems reporting well (though I'm not 100% sure). I'm testing on zigbee2mqtt 1.33

Ctrl-F4, 2023-10-09:
Read result of 'genBasic': {"hwVersion":14}
zigbee2mqtt 1.33.1-dev commit: [2aae1bd5](https://github.com/Koenkk/zigbee2mqtt/commit/2aae1bd5)
zStack3x0 Coordinator revision 20230507 (CC2652P Sonoff  ZBDongle-P)

If you need more information let me know, ready to test.
(In the meantime i have to find out how to set up OTA for convenience)

devbis, 2023-10-14:
Can you please try the latest pre-release? It is build with updated SDK and may fix reporting issues.

Ctrl-F4, 2023-10-15:
Yes it started reporting humidity with this new fw! 
I'm curious how long battery will last in comparison with the ble fw. (when configured same reporting interval) 
Thanks again!

devbis, 2023-10-15:
I don't have long-lasting results with a battery, but the expected working time should be approximately the same as with bluetooth firmware.

closed 2023-10-15 by Ctrl-F4