devbis/z03mmc #3: B1.5 Hardware LCD Support

nezg, 2023-10-15:
Thank you for this project, good job!

I have a 1.5 hardware version, everything works fine except the screen. Is it possible to make support for version 1.5? Write if you need any help.

And it would be cool if they showed the time, as in ATC firmware.

devbis, 2023-10-16:
[z03mmc.zip](https://github.com/devbis/z03mmc/files/12916780/z03mmc.zip)

Can you please try out this firmware to check if display works?

devbis, 2023-10-16:
Fixed in version [1.0.1](https://github.com/devbis/z03mmc/releases/tag/1.0.1)

closed 2023-10-16 by devbis

nezg, 2023-10-17:
Thanks, the screen is working. 
But in release 1.0.1 there is an artifact on the screen in the upper right character.
According to observations, it appears before the temperature change.
It's hard to catch, but there are on both sensors.
[image]