devbis/z03mmc #4: Support Fahrenheit degrees on display

devbis, 2023-10-16:


khayamgondal, 2023-11-04:
Hi, do you have a plan to work on this? Great initiative btw :)

devbis, 2023-11-04:
Yes, just finished it. You can try firmware from a branch build https://github.com/devbis/z03mmc/actions/runs/6756695293
Keep in mind that you should use [updated convertor for z2m ](https://github.com/devbis/z03mmc/blob/fahrenheit/converters/lywsd03mmc.js)

Will merge it later

devbis, 2023-11-09:
Added in https://github.com/devbis/z03mmc/releases/tag/1.0.6

closed 2023-11-09 by devbis

jds11111, 2023-11-09:
Does anyone know if this will show up for OTA update on Z2M, or do we need to do it manually?

Great work, devbis.

devbis, 2023-11-09:
@jds11111 You may track the PR for updates in z2m https://github.com/Koenkk/zigbee-OTA/pull/395

jds11111, 2023-11-10:
I am feeling stupid: I don't see how to set it.

devbis, 2023-11-10:
You have 4 options
1.  Add custom convertor from this repo
2. Install dev version of z2m, the updated converter is merged there
3. Write value 1 to thertmostatUIConfig/displayMode cluster manually using dev console
4. Wait until next release of z2m, the toggle will be there

jds11111, 2023-11-10:
Ahh. Thanks. For me it was hvacUserInterfaceCfg/TmpDisplayMode to 1 that did the trick.