devbis/z03mmc #6: After firmware flashing finished the sensor is down off

SAOPP, 2023-10-17:
Hi!

Guys, I think it's more a question than issue, but anyway I will ask, maybe someone got some solutions for my case... :-D

Thanks for zigbee firmware btw :+1:

I tested several of my sensors, one was flashed and is already working in my zigbee network, the second sensor, after the firmware was completed, stopped showing signs of life, it simply turned off, replacing the battery or closing the contacts produce any results. Is there any chance of restoring work through UART, maybe someone can tell me the way? Thanks.

devbis, 2023-10-17:
You can easily restore it via UART. Just follow this paragraph https://github.com/devbis/z03mmc#flashing-firmware-with-usb-to-uart

You may also refer to https://github.com/atc1441/ATC_MiThermometer/#usb-to-uart to write either BLE or Zigbee firmware.

closed 2023-10-17 by SAOPP

SAOPP, 2023-10-18:
@devbis one q, I see txd pin at board, but where is rxd? According to image of xiaomi lcd board...

SAOPP, 2023-10-18:
[image]
This photo.

devbis, 2023-10-18:
Both TX and RX are connected to the same test point. But TX with 1kohm resistor. 
[image]

SAOPP, 2023-10-18:
Ah ok, sorry, thanks :+1:

Anonym-tsk, 2023-10-18:
```
python TLSR825xComFlasher.py -p /dev/tty.usbserial-10 -b 340000 -t10000 wf 0 z03mmc.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/tty.usbserial-10, 340000 baud...
Reset module (RTS low)...
Activate (10000 ms)...
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nikolay.vasilchuk/Downloads/TLSR825xComFlasher.py", line 570, in <module>
    main()
  File "/Users/nikolay.vasilchuk/Downloads/TLSR825xComFlasher.py", line 491, in main
    activate(serialPort, args.tact)
  File "/Users/nikolay.vasilchuk/Downloads/TLSR825xComFlasher.py", line 286, in activate
    wr_usbcom_blk(serialPort, blk)
  File "/Users/nikolay.vasilchuk/Downloads/TLSR825xComFlasher.py", line 113, in wr_usbcom_blk
    return serialPort.write(blk)
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured
```

SAOPP, 2023-10-19:
@devbis Are there any differences in the boards or problematic revisions?

For example, the first photo is my board, which went down after the firmware, the second photo is Nikolai’s board. They are really different, judging by the pins on them.

[image] 
[image]