devbis/z03mmc #7: zigbee2mqtt version 1.33.1-1 isn't working

Harwesta, 2023-10-18:
zigbee2mqtt  ver. 1.33.1-1 addon in HA version not recognized this sensor after flash zigbee FW.

devbis, 2023-10-18:
Use this converter for zigbee2mqtt https://github.com/devbis/z03mmc/blob/master/converters/lywsd03mmc.js

xxddff, 2023-10-20:
I'm also using Sonoff zigbee dongle -E v2
I would like to know if there is a way to restore to the official Bluetooth firmware or pvvx custom firmware? (I want to go through OTA as much as possible instead of opening the device)

Harwesta, 2023-10-20:
> I want to go through OTA as much as possible instead of opening the device.
No chance.
Go to the light side of the force: solder the wires and flash them back

xxddff, 2023-10-20:
> > I want to go through OTA as much as possible instead of opening the device.
> 
> No chance.
> 
> Go to the light side of the force: solder the wires and flash them back
> 
> 
> 
> 

Thank you for your quick reply. I will try to open the device and flash it. I hope you can solve the EZSP coordinator issue soon. I prefer to use the zigbee network that I already have at home for these sensors, rather than bluetooth. Appreciate your excellent work :D

devbis, 2023-10-20:
[revert-to-ble-lywsd03mmc.zigbee.zip](https://github.com/devbis/z03mmc/files/13054913/revert-to-ble-lywsd03mmc.zigbee.zip)

I was lucky to use this OTA image to revert to BLE firmware again, but the other time it got bricked. You may use it but no warranty at all. It is just BLE firmware with zigbee ota header

closed 2023-10-25 by devbis