devbis/z03mmc #8: не могу подключится к другому шлюзу.

KaportsevIA, 2023-10-19:
Привет. После прошивки - один раз подключился к SLS-шлюзу. А теперь не могу перекинуть датчик на дугой шлюз. Напиши правильную последовательность перехода. В режим сопряжения. (Батарейку вытаскивал минут на 5)

Sergey-SRG, 2023-10-19:
> Привет. После прошивки - один раз подключился к SLS-шлюзу. А теперь не могу перекинуть датчик на дугой шлюз. Напиши правильную последовательность перехода. В режим сопряжения. (Батарейку вытаскивал минут на 5)

В случае, если он не включается, замкните контакты RESET и GND на плате на 3 секунды, чтобы сбросить настройки Zigbee. Замените аккумулятор, может потребоваться

dmbuligin, 2023-10-26:
Подтверждаю. 
После прошивки подключится к координатору можно только один раз. Далее подключение к другому координатору уже невозможно, замыкания RESET и GND не помогли.
Прошивка 1.0.2, Z2M
Протестировал на 3 шт.
Подготовлюсь, буду их перепрошивать через UART. По результатам отпишусь.

M0DDer, 2023-10-26:
Подтвержаю. Перепрошивка через UART опять...

valekrus, 2023-10-26:
Подтверждаю. Даже с ZHA есть такая проблема. Стойкое ощущение, что замыкание GND и RESET не даёт ничего.
Подробнее:
Изначально подключил к ZHA, датчики, как было описано в других issue периодически отваливались. Помогало, что интересно, передёрнуть батарею или замкнуть GND+RESET. Причём можно было удалить их из сети, и, при передёргивании батареи или GND+RESET они к этой же сети подключались. Решил попробовать Zigbee2MQTT, соответственно пересоздал сеть в этой системе. Другие устройства подцепились, а эти датчики никак не видит - и батарейку менял, и GND+RESET замыкал, замыкал+передёргивал батарею - вообще никак. Ок, пересоздал сеть обратно в ZHA - теперь и там их не видит больше. Похоже, что замыкание GND+RESET не сбрасывает настройки Zigbee.

M0DDer, 2023-10-29:
Прошил через UART 1.0.4. А в Z2M (перенес на другой сервер) он так и не видется....ничего не понимаю

devbis, 2023-10-29:
Наверное, перед прошивкой надо стереть флеш уголком, настройки зигби лежат в отдельном месте на флеше

devbis, 2023-10-30:
@dmbuligin @valekrus  @M0DDer Попробуйте, пожалуйста, обновиться до данной прошивки, тут есть изменения касаемо стирания настроек зигби при замыкании контактов. 

[fw.zip](https://github.com/devbis/z03mmc/files/13203316/fw.zip)

devbis, 2023-10-30:
Fixed reconnecting to another coordinator in https://github.com/devbis/z03mmc/releases/tag/1.0.5

closed 2023-10-30 by devbis