devbis/z03mmc #9: Mi home support

mgrybyk, 2023-10-19:
Hello,
Thanks a lot for your work.

Will I be able to use the device in Mi Home app?

caideone, 2023-10-20:
不支持，而且非常耗电

devbis, 2023-10-21:
MiHome is not supported yet. Not sure it is a change to implement it.

pvvx, 2023-10-27:
Из просторов инета есть прошивка ZigBee для LYWSD03MMC.
Работает с Xiaomi Gateway 3, но потребляет много.
Прикидывается 'Zigbee WSDCGQ11LM (lumi.weather) Aqara'.

https://disk.yandex.ru/d/D60QI_ZfJBh16w

Перед прошивкой стереть всю Flash, потом записать этот fw и по старту начнет регистрацию... Xiaomi Gateway 3 подтверждает регистрацию звуком.
С какими версиями HW работает не уточнял. Запустилась на B1.4 и B1.9. ( Я не испольpую Zigbee из-за большого потребления батареек при частой передаче данных и из-за малой дальности связи по сравнению с BLE Long Range. )