devbis/z03mmc #10: Датчики отваливаются

booroondook-eng, 2023-10-20:
У меня два перепрошитых датчика (был еще третий, но я его вернул на BLE).
Выясняется, что через несколько часов работы (точно не скажу - несколько часов, но не более суток) оба датчика "отваливаются" от Zigbee2MQTT. В списке устройств напротив них высвечивается красная надпись "Оффлайн".
Помогает передергивание батарейки, но с учетом конструктивных особенностей корпусов этих датчиков, да еще приклеенных на стену, передергивание батарейки превращается в сложную задачу.
Что можно предпринять, чтобы не было отвалов?

Sergey-SRG, 2023-10-20:
> У меня два перепрошитых датчика (был еще третий, но я его вернул на BLE). Выясняется, что через несколько часов работы (точно не скажу - несколько часов, но не более суток) оба датчика [...]

А что на счёт качества связи у вас? Какие показатели lqi?

SoulOnFire, 2023-10-20:
I had a power failure, all the routers and coordinators were turned off, and then only by removing the battery did they reconnect to the Z2M.

booroondook-eng, 2023-10-21:
> 
> А что на счёт качества связи у вас? Какие показатели lqi?
145 - 155

franciscofsales, 2023-10-21:
> I had a power failure, all the routers and coordinators were turned off, and then only by removing the battery did they reconnect to the Z2M.

I've had the same issue

IvanGavrilenko, 2023-10-21:
То же самое

igelbro, 2023-10-21:
Та же беда

malehovich, 2023-10-23:
+
lqi 87

Ccrrocus, 2023-10-24:
[image]

lardeg, 2023-10-24:
Тоже самое. из 5 датчиков 3 стабильно отваливаются в течении суток.

booroondook-eng, 2023-10-26:
Отвалы продолжаются - новая версия прошивки не помогла.

seregahere, 2023-10-26:
HA, Zigbee2MQTT
Прошивка 1.0.2

Днем было отключение электричества на несколько часов. После восстановления все 5 датчиков отвалились, по прошествии 2+ часов ни один не ожил. Попытки в Zigbee2MQTT из консоли разработчика считать какой-то атрибут (например кластер genBasic, атрибут manufacturerName) приводят к ошибке

> 2023-10-26 17:59:32Publish 'set' 'read' to 'temperature_guestroom' failed: 'Error: Read 0xa4c138f58fbd37ac/1 genBasic(["manufacturerName"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 48889 - 1 - 35 - 0 - 1 after 10000ms)'

lqi, как и другие данные, не обновляются с момента отключения электричества.

Mostalk, 2023-10-27:
На 1.0.4 до сих пор есть эта проблема

Ccrrocus, 2023-10-27:
[image]

Mostalk, 2023-10-27:
Просто так отвала нет, он происходит при потере связи, в любом виде. Будь то перезагрузка роутера или отдаление датчика.

Ccrrocus, 2023-10-27:
Пару раз обновлялась система с перезагрузкой, один раз выключал не надолго, вроде все норм с датчиком, менял параметры и он сразу подхватывал.

IvanGavrilenko, 2023-10-27:
[image]

Прошивка 1.0.4 - датчик стоит на одном месте,  НА не перезагружался - около 17.00 передал последние показания и сказал до свидания

Mostalk, 2023-10-27:
> Пару раз обновлялась система с перезагрузкой, один раз выключал не надолго, вроде все норм с датчиком, менял параметры и он сразу подхватывал.

Может тоже от HW зависит, но у меня два, как я понял, разных HW на датчиках. Но ситуация одинаковая. Перезагрузка HA, либо потеря сигнала - все. Нужно передёргивать батарейку

kirush0280, 2023-10-27:
Проблему подтверждаю на версии 3001-1002, если отключить электричество на шлюзе минут на 10, больше они не переподключаются.
Также могут зависнуть в процессе работы - помогает только батарейка.
Сейчас обновляется ан 3001-1004, там поменяли, как я понял только периоды опроса. Наблюдаем дальше.

devbis, 2023-10-27:
> Просто так отвала нет, он происходит при потере связи, в любом виде. Будь то перезагрузка роутера или отдаление датчика.

Спасибо за идею, да, у меня тоже воспроизводится. Буду копать как это решить.

komw, 2023-10-28:
How do you add it to the z2m? Mine says that its not supported

fryc88, 2023-10-28:
You need to add custom converter for it. It is mentioned and converter file is added, however mine gives me error regarding temperature value something

devbis, 2023-10-28:
@Mostalk @IvanGavrilenko @booroondook-eng  Can you please check out this firmware? it has some improvements regarding reconnecting but I want to double check it works with your issues

[fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

booroondook-eng, 2023-10-28:
> @booroondook-eng Can you please check out this firmware? 
> 
With firmware 1.0.3/1.0.4 disconnect problems disappeared, so I don't see the need to test the release you propose.
Thank you for your responsiveness!

Mostalk, 2023-10-28:
> [fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

I turned off the zigbee stick for a few minutes, everything seemed to be fine, the connection was restored. I'll keep an eye on it and do a couple of tests.

IvanGavrilenko, 2023-10-29:
> @Mostalk @IvanGavrilenko @booroondook-eng Can you please check out this firmware? it has some improvements regarding reconnecting but I want to double check it works with your issues
> 
> [fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

Situation has PARTLY improved - now I've obtained data for about 8 hours (instead of about 1-2)

[image]

But finally the sensor again stopped sending values (about 7.00 at the screenshot)

Alex500317, 2023-10-29:
> 
> [fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

I also tried this version. The sensors began to reconnect to the gateway after it was lost! This is very good! Thank you!

devbis, 2023-10-29:
@IvanGavrilenko 
> But finally the sensor again stopped sending values (about 7.00 at the screenshot)
> 

Did something peculiar happen on 7.00?

Mostalk, 2023-10-29:
The sensors worked for some time. Now there seem to be no problems on both sensors.

Thank you!

[image]

IvanGavrilenko, 2023-10-29:
> Did something peculiar happen on 7.00?

No h like previously just in a moment sensor stopped sending data

IvanGavrilenko, 2023-10-30:
> Did something peculiar happen on 7.00?

Now it seems all Ok - working for about 24h fine. I've flashed to the original firmware and then to the ZigBee via UART

I SUPPOSE the issue was - I've flashed the 

> [fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

file with UART, and put ota.zigbee to the ota folder and thus sensor started working as I've flashed it and when ota transfer was finished - at about 7am the sensor stopped sending data.

lardeg, 2023-10-30:
> @Mostalk @IvanGavrilenko @booroondook-eng Can you please check out this firmware? it has some improvements regarding reconnecting but I want to double check it works with your issues
> 
> [fw.zip](https://github.com/devbis/z03mmc/files/13195682/fw.zip)

Как прошить это прошивку?

devbis, 2023-10-30:
Fixed reconnecting on network loss in https://github.com/devbis/z03mmc/releases/tag/1.0.5

closed 2023-10-30 by devbis