devbis/z03mmc #11: High battery consumption

devbis, 2023-10-21:


pvvx, 2023-10-23:
With the same frequency of sending data, Zigbee has many times more consumption (!) than BLE.
The communication range in BLE Long Range mode (PHY Coded S=8) is two times longer than that of Zigbee. BLE Long Range - approximately 1 km in a straight line at TX 0 dB. The Bluetooth 5.4 standard completely overlaps ZigBee (a variant of BLE ‘advertising' PAwR, standardization of encryption of ‘advertising', ...).

Ccrrocus, 2023-10-24:
[image]

pvvx, 2023-10-24:
First run:

[image]

After binding with Xiaomi Gateway 3:

[image]

Averge: 108.9 uA

Typical impulse:

[image]

Window: 14 ms,  4.654 mA

The "Reset" contact does not reset the binding.

devbis, 2023-10-24:
@pvvx  Thanks for measurements! Reset pin waits for 3 seconds to reset Zigbee internal state to avoid accidental resetting

pvvx, 2023-10-24:
The Reset contact was held for 30 seconds...

Ccrrocus, 2023-10-24:
[image]

pvvx, 2023-10-24:
[image]

pvvx, 2023-10-29:
[Малая оптимизация по потреблению](https://github.com/pvvx/z03mmc) - среднее потребление 19 мкА для Xiaomi LYWSD03MMC B1.4, при опросе 10 сек. 

[image]

Alex500317, 2023-11-03:
> [Малая оптимизация по потреблению](https://github.com/pvvx/z03mmc) 
> 
I started testing this fork. I have two sensors on the street, one was flashed with the version from pvvx, the second from devbis. The battery charge was the same for both sensors. I'm very curious what the difference will be. Now, since the release of 1.0.4, on all my 4 sensors (2 at home and 2 outdoors), on average, the battery charge has dropped by 5-6% (after about a week of use).

pvvx, 2023-11-03:
> I started testing this fork.

Previous versions in [fork](https://github.com/pvvx/z03mmc) were intended for variant tests, not for work...
The current version  fixed to work somehow...  Measurement graph have been published.
It is not yet possible to make the consumption in the working version less than 54 µA.

Will have to test for about 4 months.
4 months is the estimated lifespan of an average CR2032 under these conditions.

Alex500317, 2023-11-03:
> > Я начал тестировать эту вилку.
> 
> Предыдущие версии в [форке](https://github.com/pvvx/z03mmc) предназначались для альтернативных тестов, а не для работы... Текущая версия исправлена, чтобы хоть как-то работать... 

Я как раз вчера увидел, что вы обновили всё, и собрал последнюю версию. К тому же расчётные 4 месяца с зигби - очень даже хорошо, на вашей ble датчики прожили уже более полугода, но у меня не всегда стабильный приём из-за особенностей квартиры. Сейчас в прошивке от devbis всё-таки высоковато потребление

devbis, 2023-11-03:
Consumption measurements shows that current version could last for 9+ months from cr2032 battery

pvvx, 2023-11-03:
> Consumption measurements shows that current version could last for 9+ months from cr2032 battery

В среднем Xiaomi LYWSD03MMC от дешевых батареек CR2032 работает от 10 месяцев. От дорогих - более года.
При этом среднее потребление не более 16 мкА (разные версии). Тест на двадцати термометрах в течении более двух лет.
Это всё варианты без конденсаторов, которые Xiaomi пожадничали поставить в предусмотренные места на печатной плату,  что сокращает срок службы батареи на 40%.

jds11111, 2023-11-05:
I am using a rechargeable battery. The new firmware has been installed only 3 days, new battery that started at 100%. It is now down to 86%.  Loses about 5% a day. Sorry if this is irrelevant---I can't read many of these replies.

devbis, 2023-11-05:
@jds11111  Battery remaining percentage is not linear. It would be nice if you could report about how many days your rechargeable battery will last until 0%

pvvx, 2023-11-05:
[image]

https://habr.com/ru/companies/lamptest/articles/555158/

Differences in CR2032 battery capacities  - 11...100% 

@devbis - The range 0..100% is plotted according to the formula from the [source code](https://github.com/devbis/z03mmc/blob/master/src/device.c#L193C8-L193C8):
```
#define BATTERY_SAFETY_THRESHOLD 2200 // mV

    voltage = drv_get_adc_data(); // mV
	percentage = ((voltage - BATTERY_SAFETY_THRESHOLD) / 4);
	if (percentage > 0xc8) percentage = 0xc8;
```
[image]

@jds11111  - Given the incorrect calculations of the remaining capacity in the current code, the real remainder of your battery is about 10..20%.

devbis, 2023-11-06:
@pvvx Your calculations are wrong. 100% is at 3.0v

pvvx, 2023-11-06:
Yes, I forgot that Zigbee is all weird and not suitable for people. The current battery percentage in Zigbee is 200%. :)

pvvx, 2023-11-09:
Release 1.0.6

Rejoinig cycles:

[image]
The battery will only last 2 weeks. It is necessary to reduce energy consumption to operate for a month on an average CR2032 battery.

Press key 3 sec and Join:

[image]

Pasue, In between (re)joinig (Only sensor polling and display):

[image]
Lack of optimization.

andrazek, 2023-11-13:
Reducing battery consumption should be top priority. Is it possible to make it as efficient as AQARA WSDCGQ11LM, which has 2 years of battery life with the same CR2032 battery?

devbis, 2023-11-13:
@andrazek It probably could. But this sensor has a different philosophy. While other sensors can sleep longer and wake up to send data every 5 minutes, this sensor needs to update screen in reasonable period. For this device it is 10 seconds. 

You can build your own fork: disable screen module and significantly increase coordinator poll period and sensor read period to increase battery life.

vladimir1408v, 2023-11-13:
[image]
Заряд батареи сильно расходуется за сутки, почти на 10% прошивка 1.6 как это исправить?

devbis, 2023-11-13:
@vladimir1408v You may revert to bluetooth firmware using https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota and local OTA index in zigbee2mqtt.
Battery drain is not linear and one day is not enough to say how long it will work.

vladimir1408v, 2023-11-13:
> @vladimir1408v You may revert to bluetooth firmware using https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota and local OTA index in zigbee2mqtt. Battery drain is not linear and one day is not enough to say how long it will work.

Спасибо за ответ и за проделанную работу, я останусь на вашей версии, и понаблюдаю. Мне кажется это началось на 1.6 до этого была 1.5 и расход был меньше, по ощущениям возможно я не прав

devbis, 2023-11-13:
@vladimir1408v Sometimes, you need to rejoin the sensor after OTA to get lower consumption rate. I was told that sometimes after OTA it polls more frequently than during normal work.

pvvx, 2023-11-13:
Extremely high consumption is observed if the network coordinator is disabled.

breaver, 2023-11-25:
> @vladimir1408v You may revert to bluetooth firmware using https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota and local OTA index in zigbee2mqtt. Battery drain is not linear and one day is not enough to say how long it will work.

Thanks for that hint. I have 10 sensors which i am planning to convert to Zigbee using your firmware. Glad to hear that there is an easier way back to a bluetooth based firmware in case the power consumption is to high (6 months would be good). 
Do you have an example for the local index file ? (manufacturer code, etc.)

pvvx, 2023-11-25:
> Do you have an example for the local index file ? (manufacturer code, etc.)

File name decryption?
Sample: "1141-020a-01103001-Z03MMC.zigbee" 

| Manufacturer Code | Image Type | File Version | Stack Version | Name | Ext OTA |
| -- | -- | -- | -- | -- | -- |
| 1141 | 020a | 0110 | 3001 | Z03MMC | zigbee |
| 0x1141 - Telink | 0x02 - TLSR825x, 0x0a - Xiaomi LYWSD03MMC | App release 0.1, App build 1.0 | Zigbee v3.0, Release 0.1 |  Z03MMC | OTA |

|Image Type| Device, note |
| -- | -- |
| 0x0201 |  MHO-C401 (old version) |
| 0x0202 |  CGG1 (old version) |
| 0x0203 |  LYWSD03MMC ver https://github.com/devbis/z03mmc |
| 0x0204 |  WATERMETER ver https://github.com/slacky1965/watermeter_zed |
| 0x0206 |  CGDK2 |
| 0x0207 |  CGG1 (new version) |
| 0x0208 |  MHO-C401 (new version) |
| 0x0209 |  MJWSD05MMC |
| 0x020A |  LYWSD03MMC ver https://github.com/pvvx/ZigbeeTLc |
| 0x020B |  MHO-C122 |
| 0x0210 |  Water Tank sensors |
| 0x0211 |  TS0201-TZ3000 |
| 0x0212 |  TS0202-TZ3000 |

andrazek, 2023-11-26:
> @andrazek It probably could. But this sensor has a different philosophy. While other sensors can sleep longer and wake up to send data every 5 minutes, this sensor needs to update screen in [...]

Thanks for clarification. Still, why would consume more on Zigbee compared to bluetooth, if both having the same update frequency? I always thought zigbee is the most efficient tech for battery devices. Would extending the reading interval to every 30s instead of 10s noticeable improve the battery life? Having a lot of these devices and short battery life is pain in the ass.

pvvx, 2023-11-26:
> I always thought zigbee is the most efficient tech for battery devices. 

In BLE mode, with default settings, the screen refresh interval is 2.5 seconds. Data transmission via RF every 2.5 seconds on three channels. Sensor polling - 10 seconds.
When switching to LE Long Range mode (communication range of 1 km for RF transmit power 0 dB), battery consumption increases. The time for transmitting and receiving packets increases. But you can increase the transmission interval to 3..5 seconds for an equivalent level of battery consumption.

jds11111, 2023-11-26:
An update on power consumption in my hygrometer. I am using a rechargeable battery, which has its own power curve. Within a couple of days it dropped to about 83%. I did update the firmware about that time, but the drain rate could just be the battery. It has now been about four weeks and is draining much more slowly. Unfortunately, HA is storing data on the device for only a couple of weeks. Here is the last 10 days to give an idea. I assume that the fluctuations are the temperature swings in the house. I would estimate the current rate to be about 0.4%/day. 

[image]

jds11111, 2023-11-26:
Yeah, the oscillations in battery are definitely the temperature:

[image]

jds11111, 2023-11-27:
>  There is no point in analyzing consumption based on the behavior of a specific battery.

I would argue that a single data point is more useful than zero data points.

pvvx, 2023-11-29:
[Dual zigbee--ble device]( https://github.com/pvvx/BZdevice/blob/master/README.md#dual-zigbee--ble-device) - Xiaomi LYWSD03MMC B1.4 - 18 µA

jds11111, 2023-11-29:
Do I understand this correctly? You are running both radios for simultaneous ZigBee and BLE, with just 18uA?? That is wild.

pvvx, 2023-11-29:
> Do I understand this correctly? You are running both radios for simultaneous ZigBee and BLE, with just 18uA?? That is wild.

Yes, this is a big battery drain.

Cycle = 5 sec = 5000 ms
Sleep current = 6 µA = 0.0006 mA
The duration of the SoC activity and transmission cycle varies depending on the Zigbee report settings. 
If we take an average of 10 ms and 6 mA, then we get:
(4990ms*0.006mA+10ms*6mA)/5000ms  = 0.017988 mA = 18 µA

pvvx, 2023-11-29:
Dual zigbee-ble device, RF TX +3dB

[image]

[image]

[image]

[image]

If there are no additional Zigbee polls:
```
((10000-9-26)*0.006+9*3.7+26*3.99)/10000=0.019683mA
```

Bolukan, 2023-11-29:
I would benchmark against 1 year endurance, a CR2032 gives 240mAh (220-240 mAh is common presumed). With these numbers the budget would be 27,4 µA of which 6 µA seems to be consumed for the continuous sleep mode. The latest configuration (poll+lcd+ble+zigbee) would just fit (97% of budget)

pvvx, 2023-11-29:
Only a fresh CR2032 from the best manufacturers produces 240 mAh at a constant test load with a current of less than 0.5 mA.
With a pulsed load, the final power will be 60% of the rated power.
Commercially available CR2032 have capacities ranging from 20% to 100%.

Ccrrocus, 2023-12-05:
[image]
Установил для пробы параллельно CR2032 электролитический конденсатор 200µF*10V. Батарея, как видно на графике имела заряд около 15%. Конденсатор припаял на проводах к имеющимся контактам на плате. Там, где предполагалась его штатная установка. SMD вариант сходу не нашел, но теста это не важно. Результат меня удивил - заряд батареи поднялся до 45%, притом он стабилен. Интересно, сколько протянет.

Ccrrocus, 2023-12-05:
[image]
Теперь я взял дохлую батарею, с которой уже термометр не работает. Подпаял аналогичный конденсатор, но вдвое большей ёмкостью 470µF. И прибор заработал! Понятно, что чудес не бывает и и заряд батареи сильно просел, но он работает до сих пор!

pvvx, 2023-12-05:
The larger the capacitance of the capacitor, the greater its leakage current. Balance required.

Christian-fhs, 2024-01-13:
> Just to comment that in my case the battery completely drained after I flashed the device with the zigbee firmware in less than one week. It was a new battery.

After flashing custom firmware onto five LYWSD03MMC devices, I observed a complete battery drain within just two weeks. While my initial assumption pointed towards low-quality stock batteries, upon further reflection, it seems more likely that the issue lies with the Zigbee firmware.

wf2020, 2024-01-14:
How about fitting in a CR2450? Some other from the shelf Zigbee thermometers seem to use one. From first sight it might be possible to just 3d print a new PCB cover so a CR2450 possibly fits in.

pvvx, 2024-02-11:
Firmware version 1.1.0.0,  Xiaomi LYWSD03MMC HW: B 1.4.
With 4 reports in 8 minutes:
[image]
Average 29.3 uA.

gryzli133, 2025-01-19:
I have switched to BLE 5.0 with coded PHY and have a feeling, that since that the battery life is shorter, only few months with CR2032. Do I need to change the default values to increase the battery life? Or is there anything I'm doing wrong?

pvvx, 2025-01-19:
In LE LR (Coded PHY) the consumption increases by about 3..4 times. Not suitable for CR2032.
CR2450 in LE LR with a period of 5 seconds lasts 2 years.

pvvx, 2025-01-19:
> only few months with CR2032.

CR2032 Batteries Test (best and worst battery on the market):

[image]

Load current: start 3 mA, stop 2 mA

https://batterytest.ru/

pvvx, 2026-01-05:
It's important to keep in mind that different HW versions have different battery consumption, which can vary by up to 30%.
The newer the version, the higher the consumption.

#### HW versions LYWSD03MMC

|HW | LCD I2C   addr | SHTxxx   I2C addr | Note |
|-- | -- | -- | -- |
|B1.4 | 0x3C | 0x70   (SHTC3) |  |
|B1.5 | UART | 0x70   (SHTC3) |  |
|B1.6 | UART | 0x44   (SHT4x) |  |
|B1.7 | 0x3C | 0x44   (SHT4x) | Test   original string HW |
|B1.9 | 0x3E | 0x44   (SHT4x) |  |
|B2.0 | 0x3C | 0x44   (SHT4x) | Test   original string HW |
|PCB Mark: B1.6* (2025) | SPI | 0x4x   (SHT4x/[GXHT40](https://github.com/user-attachments/files/24424893/CZ_SHT40-ad1b_GX_GXCAS_0001.pdf)/other clones...) | Custom HW: B1.1  |
|PCB Mark: B1.5* (2025) | SPI | 0x7x   (SHTC3/GXHTC3/other clones...) | Custom HW: B1.1 |

* "PCB Mark: B1.6/B1.5" are new variants that went on sale in 2025. These devices use old printed circuit boards from warehouses. They are equipped with clones of cheaper sensors and an SPI-controlled LCD controller.