devbis/z03mmc #12: EZSP support

devbis, 2023-10-21:
Some users report that it doesn't work with EZSP coordinators. Need to investigate

Mostalk, 2023-10-22:
Coordinator type: EZSP v12
Coordinator revision: 7.3.1.0 build 176

Work correctly with external converter

```
{
    "battery": 58.5,
    "humidity": 53,
    "last_seen": "2023-10-22T03:51:28.551Z",
    "linkquality": 244,
    "temperature": 23.6,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": "idle"
    },
    "voltage": 2600,
    "update_available": null
}
```

xxddff, 2023-10-22:
After resetting the device and rejoining the zigbee network, my Sonoff zigbee dongle-E v2
(Coordinator type: EZSP v8. Coordinator revision: 6.10.3.0 build 297)
It is now running normally and obtaining data stably.
In addition, the temperature and humidity Min rep change Actions in Reporting is 100, which will cause the sensor to rarely actively report. I turned it down to suit my environment. Not sure if this is a question

devbis, 2023-10-22:
100 means 1 unit, degree or percent. (it is stored in 0.01). You may change temp value to 20 to report every 0.2 degrees. For humidity the sensor resolution is 1%, so no matter to decrease it.

Mostalk, 2023-10-22:
Hmm, i new in Zigbee, but i change update time in "Min rep interval" to any value, like 1. And get realtime updating, but this setting makes so many power consumption

Mostalk, 2023-10-22:
Look at this.  Where battery on 75% it's ble firmware, after install zigbee firmware battery drop to 59%.

On chart you can see where i set update interval on 1 sec. To much power consumption. Letter i set interval to 60 sec. In my opinion in this mode battery can work so long 

[image]

Alex500317, 2023-10-22:
With firmware 1.0.2, the sensors also fall off, this happens when the signal from the gateway becomes unstable, or the gateway has been rebooted. The only way to fix it is by removing/inserting the battery. Gateway - Xiaomi Multimode Gateway 2 (ZNDMWG04LM), everything is connected via XiaomiGateway3 and Z2M

devbis, 2023-10-30:
Fixed reconnecting to another coordinator in https://github.com/devbis/z03mmc/releases/tag/1.0.5

malehovich, 2023-11-03:
После смены на Sonoff zigbee dongle-E v2  "product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8" на последней прошивке устройство не регистрируется. По логам оно как будто вообще ничего не шлет.  На dongle-P все работает. Прошивал через OTA на dongle-P. Рядом датчик Xiaomi на dongle-E регистрируется без проблем. Батарейку вынимал. Контакты GND и reset замыкал.

closed 2024-02-07 by devbis