devbis/z03mmc #13: OTA upgrade to new version doesn't work

xriser, 2023-10-22:
I see a new version has been released (1.0.2)
Is it possible to upgrade via OTA 1.0.1 -> 1.0.2?

Currently, it doesn't work in z2mqtt.

`Zigbee2MQTT:info  2023-10-22 14:59:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0xaxxx"},"error":"Failed to check if update available for '0xaxxx' (Device didn't respond to OTA request)","status":"error","transaction":"co4wv-10"}'
Zigbee2MQTT:error 2023-10-22 14:59:41: Failed to check if update available for '0xaxxx' (Device didn't respond to OTA request)
Zigbee2MQTT:debug 2023-10-22 14:59:41: Error: Device didn't respond to OTA request`

devbis, 2023-10-22:
Did you follow this instruction? https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

It works for me using this manual and .zigbee OTA image from the release.

glinkin55, 2023-10-22:
Please write a short instruction for us

xriser, 2023-10-22:
Thank you very much.
I've followed the instructions and created a custom JSON file for OTA.
Devices has been successfully updated.

short instruction:
add 
```
ota:
  zigbee_ota_override_index_location: /app/data/ota.json
```
to configuration.yaml

create ota.json, example:
`[
  {
        "fileVersion": 268578817,
        "fileSize": 129378,
        "manufacturerCode": 4417,
        "imageType": 515,
        "sha512": "5604d795ff9a88c88a6ddbc58f11178247e89629f14412448c1daba11f3079c43e14ce779fa328074bde1ed852648051a72f2a9be62bd637abf19cf5d78875f7",
        "url": "https://github.com/devbis/z03mmc/releases/download/1.0.2/1141-0203-10023001-z03mmc.zigbee"
  }
]`

updrade via z2mqtt OTA menu

closed 2023-10-22 by xriser