devbis/z03mmc #14: ZHA OTA?

sneakythr0ws, 2023-10-22:
Hi!
Help me pls
I tried to `OTA` update the device via [ZHA](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates)
I just put z03mmc.bin in `otau_directory` and turned on debug log
As I understand it linked `z03mmc.bin` and `LYWSD03MMC` device.
But OTA update doesn't start
```
2023-10-22 18:48:04.297 DEBUG (MainThread) [zigpy.zcl] [0xF3D3:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=4417, image_type=515, current_file_version=268513281, hardware_version=None, model='LYWSD03MMC'
2023-10-22 18:48:04.298 DEBUG (MainThread) [zigpy.zcl] [0xF3D3:1:0x0019] No OTA image is available
```
May be I have to rename file?

devbis, 2023-10-22:
For OTA you should use .zigbee file, it is a firmware with OTA header. It is located shopping with .bin file

sneakythr0ws, 2023-10-22:
> For OTA you should use .zigbee file, it is a firmware with OTA header. It is located along with .bin file in release files

Thank you! It works fine via [instruction](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates)

closed 2023-10-22 by sneakythr0ws

Timmiej93, 2023-10-29:
@sneakythr0ws Did you have to do anything special? I'm trying to do the same thing, but I'm still getting the "No OTA image is available" error with the `.zigbee` file in my `otau_directory`.

sneakythr0ws, 2023-10-29:
> @sneakythr0ws Did you have to do anything special? I'm trying to do the same thing, but I'm still getting the "No OTA image is available" error with the `.zigbee` file in my `otau_directory`.

Nothing special. Did you try to reboot the system after it?

Timmiej93, 2023-10-29:
I've tried rebooting, reloading yaml, even different `otau_directory`s. Could you confirm that this is (roughly) the yaml you have in your configuration file?
```
zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: true
```

Timmiej93, 2023-10-29:
Odd, I just kept trying again and it suddenly started upgrading. 🤷‍♂️

Just to be clear, for anyone who may run into this in the future, do you need both the `.zigbee` file and the `z03mmc.bin` firmware in the ota folder?

sneakythr0ws, 2023-10-29:
> Odd, I just kept trying again and it suddenly started upgrading. 🤷‍♂️
> 
> Just to be clear, for anyone who may run into this in the future, do you need both the `.zigbee` file and the `z03mmc.bin` firmware in the ota folder?

👍 No. You need .zigbee only