devbis/z03mmc #15: How to upgrade to new release

booroondook-eng, 2023-10-22:
My sensors have ZB firmware 1.0.1 now.
How can I upgrade these sensors to new release (1.0.2) without using the invasive method of soldering wires, UART connection etc.

sneakythr0ws, 2023-10-22:
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

booroondook-eng, 2023-10-22:
> https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

Have you tried updating LYWSD03MMC by yourself using this method?
(I'm guessing that not only have you not tried this, but you haven't even read the description of this method  :-) ).

sneakythr0ws, 2023-10-22:
> > https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates
> 
> Have you tried updating LYWSD03MMC by yourself using this method? (I'm guessing that not only have you not tried this, but you haven't even read the description of this method :-) ).

Don't judge by yourself. I successfully updated from 1.0.1 to 1.0.2

booroondook-eng, 2023-10-22:
I have no ZHA but use Zigbee2MQTT

booroondook-eng, 2023-10-22:
1. Go to Zigbee2MQTT directory which contains configuration.yaml file
2. Place OTA-upgrade file (e.g.1141-0203-10023001-z03mmc.zigbee) downloaded from this repository to the above mentioned directory 
3. Add the folliwing lines to configuration.yaml:
```
ota: 
    zigbee_ota_override_index_location: my_index.json 

```
(you can use any file name instead of my_index.json))
4. Create my_index.json (or your chosen filename) and add the folloowing strings to it:
```
[
    {
        "url": "1141-0203-10023001-z03mmc.zigbee"
    }
]
```
(Use the actual file name)
5. Restart Zigbee2MQTT from its web interface
6. Go to Zigbee2MQTT "OTA" Menu
7. Check first and the update your sensors. The process takes about 20 minutes or even more.

SoulOnFire, 2023-10-23:
I just update to new version using Zigbee2MQTT using the OTA menu without any issue.

closed 2023-10-23 by booroondook-eng

kolobok22, 2023-10-23:
I have SLS gateway and mosquitto broker - how can I do the upgrade procedure?