devbis/z03mmc #17: Data rounded to the nearest hundredth.

Nab0y, 2023-10-22:
Is it possible for the temperature and humidity data to be rounded to the nearest hundredth of a percent? For example, not 21.3, but 21.38 and so on.

Mostalk, 2023-10-22:
In my case humidity only in integer. Where float values on chart it's BLE firmware, then Zigbee. 

[image]

I try changing this values, but nothing happened 

[image]

Temperature always have one value after the decimal point, but humidity doesn’t have this value at all

```
{
    "battery": 61,
    "humidity": 56,
    "last_seen": "2023-10-22T20:59:53.477Z",
    "linkquality": 200,
    "temperature": 23.5,
    "update": {
        "installed_version": 268578817,
        "latest_version": 268578817,
        "state": "idle"
    },
    "voltage": 2600
}

```

devbis, 2023-10-26:
It is fixed in release 1.0.3

Mostalk, 2023-10-27:
Great work. Thanks.

closed 2023-10-28 by devbis