devbis/z03mmc #18: Add setting to change TX antenna power

Mostalk, 2023-10-22:
Can you add ability to change antenna power in Z2M? 

I have one sensor at a fairly large distance from the Zigbee stick and some data loosing. It would be a good idea to manually control the TX antenna power within reasonable limits

blinkzz, 2023-10-23:
Would like to see the same for ZHA by exposing some number input or range slider, after upgrading from 1.0.1 to 1.0.2 some sensors fail to deliver their status now

pvvx, 2023-10-23:
Without hardware improvements, it is impossible to increase the output of the transmitter. There is not enough CR2032 battery current. It will only work for the first few days on a fresh battery. Further, the internal resistance of the battery will increase and instead of increasing the recoil, there will be a decrease due to voltage drawdown during transmission.

Mostalk, 2023-10-23:
@pvvx, I assumed that it was possible to change the power TX antenna since the @devbis decreased the TX antenna power to 3 dBm in the latest release. I don’t know what the power was before, but my sensor, which I mentioned above, worked better at the previous power value

pvvx, 2023-10-23:
+3 dB is the limit for Xiaomi LYWSD03MMC with CR2032. This already requires the installation of a capacitor in the place provided by the developers on the printed circuit board.
The following power steps (up to +10 dB) require a different battery...  The peak current (TX) will be more than 25 mA, which will lead to a drawdown of the CR2032 below 2 volts.

Mostalk, 2023-10-23:
Now I understand that this is why the battery was very drained on the previous firmware version. Judging by the file changes, the power was not specified at all. That is, the sensor tried to operate at a maximum of 10 dB?

pvvx, 2023-10-23:
No.
Special switching from the program is required. After +3 dB, the chip switches the transmitter power directly to the battery, without the built-in DC-DC. As a result, with an increase in power, the supply voltage drops sharply, and the return on the CR2032 becomes even less than at -10..-20 dB :)

zzz676, 2023-10-24:
> No. Special switching from the program is required. After +3 dB, the chip switches the transmitter power directly to the battery, without the built-in DC-DC (1.8V). As a result, with an increase in power, the supply voltage drops sharply, and the return on the CR2032 becomes even less than at -10..-20 dB :)
> 
> A half-discharged CR2032 battery has an internal resistance of 100 ohms. The thermometer must operate from a 3 V source with a series resistance of 100 ohms, otherwise the battery will be used up by 50% (to the dumpster).
> 
> The optimal operating mode for Xiaomi LYWSD03MMC (without circuit additions) from CR2032 is +0 dB (<8mA p-p). Less or more will be worse, because Zigbee receivers have poor sensitivity compared to BLE(5.0+). And there is no point in less return due to the total consumption of the CPU itself. Use a gateway or [ZigBee Direct](https://csa-iot.org/all-solutions/zigbee/zigbee-direct/). [Telink Zigbee-Direct](http://wiki.telink-semi.cn/wiki/protocols/Zigbee/#introduction-of-zigbee-direct) or Telink SDK ['Zigbee + Bluetooth LE Concurrent Mode'](http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/#software-development-kit)

А что если дать возможность снижать мощность передатчика, для экономии батареи. У меня, допустим, эти датчики стоят в 50 см от выключателей с зигби, работающих как роутер. И возможно и с меньшей мощностью антенны нормально бы передавали информацию.

pvvx, 2023-10-24:
В текущей реализации передача (при +0 дБ) дает потребление на длительность посылки самой передачи где-то в 8 мА. При этом основное время в ZigBee уходит на прием ответа обработку "стека". И CPU ещё разогнан на 48MHz, что дает потребление в этих режимах 5..6 мА. Т.е. вы хотите немного убрать горбик передачи, вылезающий среди общей активности во время периода пробуждения SoC на пару мА? Т.е. ужать общее потребление на пару процентов?
См. текущие [диаграммы](https://github.com/devbis/z03mmc/issues/11#issuecomment-1776982827).

devbis, 2023-10-25:
В моей прошивке частота выставлена в 24 МГц. 

[image]

По графику зависимости мощности от тока видно что где-то после 2.5 dBm  потребление начинает расти круче.

Mostalk, 2023-10-25:
> По графику зависимости мощности от тока видно что где-то после 2.5 dBm потребление начинает расти круче.

Хмм, интересно, почему между 2.5 и 3.5 такой сильный скачек, а потом идет более гладко

zzz676, 2023-10-25:
> В моей прошивке частота выставлена в 24 МГц.
> 
> [image]
> 
> По графику зависимости мощности от тока видно что где-то после 2.5 dBm потребление начинает расти круче.

Судя по графику может имеет место и сделать 2.5 (до редкого повышения)

devbis, 2023-10-25:
Да, уже снизил до 2dBm, чтобы с небольшим запасом. В будущей версии будет так.

pvvx, 2023-10-25:
Этот график не соответствует пиковому потреблению при передаче (работе передатчика).
В LYWSD03MMC расчетные разработчиками и рекомендуемые производителем чипов конденсаторы в питание Xiaomi не впаял - пожадничал милли-доллар - оставив пустые позиции на плате.

При RF TX 0 дБ ток 8+ мА, напряжение полу-разряженной CR2032 падает до 3-100*0.008=2.2В
При +3дБ падение напряжения уже приводит к перезагрузке или зависанию чипа.

pvvx, 2023-10-25:
> Хмм, интересно, почему между 2.5 и 3.5 такой сильный скачек, а потом идет более гладко

Потому, что на уровнях +3дБ производится переключение питания передатчика напрямую к внешнему питанию, минуя встроенный в чип DC-DC. 
Уровень выходной мощности в SDK задается специальными значениями, в которых некоторые биты указывают к какому питанию будет подключен передатчик и возможны разные комбинации на уровнях 0..+3дБ.
График показывает потребление только части RF, без учета потребления CPU и остальной включенной периферии в чипе, да в каких-то идеальных условиях.
Реальные показатели тока при передаче на уровне +10дБ уходят далеко за 25 мА.

aurbod, 2024-08-19:
I have a follow up question: I'd like to power my sensors via a power supply: a central dc-dc converter providing 3.3v where all my sensors are attached too with cable length < 10m. In addition each sensor will get a 100nF ceramic capacitor locally (which should add as a buffer?). My question is: does anybody see any problems with this approach or should it work? And how do I change the transmit power? From my understanding I should change **g_zb_txPowerSet ** in _src/common/main.c)_:

`int main(void){
    g_zb_txPowerSet = RF_POWER_INDEX_P1p99dBm;`

But I can't find those constants documented anywhere? What would be possible values? And what would be the max supported output power (+10dB)? Thanks for your answers!

pvvx, 2024-08-19:
> if I understand this correctly, setting g_zb_txPowerSet to RF_POWER_INDEX_P10p46dBm would mean the TX power would be +10.46 dbm?

Yes.

aurbod, 2024-08-20:
Ok thanks for your quick reply. Do you have any insights as to whether powering the sensors via a wired power supply and with a local buffering capacitor should work or if there could be any problems?

pvvx, 2024-08-20:
The voltage of the source should not exceed 3.6V.
The resistance of the wires should not be more than 100 Ohms. 
The rest is not particularly important.

aurbod, 2024-08-28:
Ok, so I've implemented the changes, here are the results:
Before I had a Linkquality lqi of around 30 on the floor above my zigbee stick with the original devbis firmware
After setting RF_POWER_INDEX_P10p46dBm  the lqi doubled to at least 65
I'm happy, this should work fine for my needs. Note that there are only a few meters between the two floors with only wood in between, so your z03mmcs probably won't cover more than 2 floors and/or concrete floors. I for example did not get a signal in the cellar probably because the humidity is higher?

pvvx, 2024-08-28:
It is advisable to try changing the direction of the antenna (thermometer body and receiving antenna). They all had directionality. Rotating the receiver with an internal antenna often gives an increase of more than +10 dBm.