devbis/z03mmc #20: Two-stage firmware?

pvvx, 2023-10-23:
If the size of the fw bin file is less than 128 kilobytes, then two-stage firmware from the original (MiHome) is not required. It is necessary to enter the code in the cold start to move the fw to the required area of flash memory. There is an example in the firmware for MJWSD05MMC.
TelinkMiFlasher and some custom firmware already support advanced OTA for writing files larger than 128 kilobytes to the 0x40000 area.
Zigbee also has the ability to restore the original (original from MiHome) firmware via OTA.

Если размер файла fw bin меньше 128 килобайт, то двухэтапная прошивка из оригинала (MiHome) не требуется. Необходимо вписать код в холодный запуск для перемещения fw в требуемую область флэш-памяти. В прошивке для MJWSD05MMC есть пример.
TelinkMiFlasher и некоторые пользовательские прошивки уже поддерживают расширенный OTA для записи файлов размером более 128 килобайт в область 0x40000.
Zigbee также имеет возможность восстановить исходную (оригинал от MiHome) прошивку через OTA.

devbis, 2023-10-25:
The size for the firmware is almost 128kb that makes impossible to flash it with you flashing tool after adding more features and spilling over 129kb (due to restrictions in the web UI). In this case bluetooth firmware should be able to write a firmware to 0x40000 address, what is intermediate firmware does. 

ota_40000 is literally doing what you've said (it's your code modified). It looks for the boot flag in 0x20000, moves itself to 0x40000, and changes its segments to let next zigbee firmware be written to 0 or 0x40000.

pvvx, 2023-10-25:
For double flashing, you need to use a random MAC.
Otherwise, there will be problems described [in this issue](https://github.com/devis/z03mmc/issues/22#issuecomment-1778861095 )

Для двойной перепрошивки требуется использовать случайный MAC.
Иначе будут проблемы, описанные [в этом issue](https://github.com/devbis/z03mmc/issues/22#issuecomment-1778861095)

pvvx, 2023-11-12:
**Starting from firmware version v4.6, all thermometers in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) (v6.9) support BigOTA.**

BigOTA is enabled if the OTA file is larger than 128 kilobytes.
BigOTA download occurs at address 0x40000.
BigOTA size limit is 221184 bytes. The MAC of the device is written at address 0x76000.

If the battery is low, the connection may be interrupted. Then you will need to download again.
For example, MiHome does not allow OTA when the battery charge is less than 50%.

---

**Начиная с версии прошивки v4.6, все термометры в [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) (v6.9) поддерживают BigOTA.**

BigOTA включается, если файл OTA более 128 килобайт.
Загрузка BigOTA происходит по адресу 0x40000.
Ограничение размера BigOTA в 221184 байт. По адресу 0x76000 записан MAC устройства.

При слабой батарее возможен разрыв связи. Тогда потребуется повторить загрузку. 
Например, MiHome не разрешает ОТА при заряде батареи менее 50%.