devbis/z03mmc #21: Slow down the internal timer

Nab0y, 2023-10-24:
I suggest slowing down the internal timer by a factor of six. Personally, it won't make much of a difference for me whether the screen refreshes every 10 seconds or every minute. I don't experience sharp temperature fluctuations, and as a short practice has shown, the sensors don't react that quickly either. The benefits of this approach are much greater: fewer sensor queries, fewer value calculations, screen updates, and, as a result, reduced energy consumption.

I've assembled a test version for myself, and while I only have limited statistics for half a day, the battery consumption graph already looks more optimistic than it did before the changes.

Предлагаю замедлить внутренний таймер в шесть раз. Лично для меня не большая разница будет экран обновляться раз в 10 секунд или раз в минуту, таких резких перепадов температуры у меня нет, да и как показала недолгая практика сами датчики так быстро не реагируют. А пользы от такого решения намного больше, меньше опросов датчиков, меньше расчетов значений, обновлений экрана, как следствие меньше потребление энергии.
Я собрал себе тестовую версию, статистики пока мало всего пол суток, но график потребления батареи уже выглядит более оптимистично, чем было до изменений. 
[image]
[image]

My report settings are as follows
[image]

pvvx, 2023-10-24:
10 секунд иногда уже много, для возможности удерживать температуру с гистерезисом в 0.1С, а также для любых PID регуляторов.

Nab0y, 2023-10-28:
@devbis After the update 1.0.4, energy consumption has been normalized, and I think we can close this task as unnecessary.

closed 2023-10-28 by Nab0y