devbis/z03mmc #22: Ошибка при прошивке z03mmc.bin после промежуточной

kachalovmg, 2023-10-24:
Промежуточной прошивается нормально, вытаскиваю батарею, обновляю страницу. Подключаюсь повторно для прошивки zigbee после нажатия на Start Flashing, начинается прошивка, статус 0% и почти сразу дисконет. 
Пробовал на двух датчиках, результат одинаков.
20:41:47: Searching for devices
20:42:18: Connecting to: ATC_BA09F9
20:42:21: Hardware Revision String: B1.5
20:42:21: Software Revision String: V4.4
20:42:21: Firmware Revision String: github.com/pvvx
20:42:21: Detected custom Firmware
20:42:21: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.4, Sensor: SHTC3 (SHTV3)
20:42:21: Custom config HEX string: 55051000002804a931318ab4
20:42:43: File: z03mmc.bin
20:42:43: File size: 129316 bytes
20:42:43: Count: 8083
20:42:47: Start DFU
20:42:53: Update error: NotSupportedError: GATT operation failed for unknown reason.
20:42:53: Disconnected.

booroondook-eng, 2023-10-24:
Попробуйте выполнить эту операцию не на ПК, а на смартфоне (только не забудьте файл прошивки предварительно поместить на смартфон). Все точно так же - через браузер Chrome.

pvvx, 2023-10-24:
Тут может влиять проблема, если ранее была зашита прошивка Zigbee. Она устанавливает защиту  записи на Flash и OTA в область нулевых адресов Flash штатным SDK Telink невозможна.

kachalovmg, 2023-10-25:
Датчики оба новые, только из коробки. Третий лежит ждёт.
Попробую с телефона, может что то с блютуз ноута его не устраивает. На спрутаи видел человека с такой же ошибкой.

pvvx, 2023-10-25:
Есть ещё такое:
После смены прошивки с оф. на "кастом" требуется перезагрузить BT адаптер, а не термометр. Т.е. полностью комп.
BT адаптер и система запоминают все UUID устройства привязав их к MAC устройства и работают некорректно, пока не обновят данные.
В некоторых случаях, при разных драйверах BT адаптера, очень сложно заставить систему забыть старые данные, фиксированные по MAC устройства. В таких случаях помогает смена MAC устройства...
На Linux желательно стереть память о устройствах в '/var/lib/bluetooth/' .

kachalovmg, 2023-10-25:
Перезагрузка ноутбука, очистка кеша chrome не помогла, прошивка с другого устройства, так же нет результата.

zzz676, 2023-10-27:
> Где вы берете такие старые термометры - Hardware Revision String: B1.5 (производились более двух лет назад)? B1.5 наверно имеет самое большое потребление из всех из-за своего набора микросхем.
> 
> Идущие с ними батарейки наверно уже сели от старости и неправильных условий хранения (некоторые CR2032 просто умирают в ноль при перевозке зимой по России).
> 
> The version is determined by the addresses of the display and sensor on I2C
> 
> HW	LCD I2C addr	SHTxxx I2C addr	Note
> B1.4	0x3C	0x70 (SHTC3)	 
> B1.5	UART!	0x70 (SHTC3)	 
> B1.6	UART!	0x44 (SHT4x)	 
> B1.7	0x3C	0x44 (SHT4x)	Test original string HW
> B1.9	0x3E	0x44 (SHT4x)	 
> B2.0	0x3C	0x44 (SHT4x)	Test original string HW
> Version 1.7 or 2.0 is determined at first run by reading the HW line written in Flash. Display matrices or controllers are different for all versions, except B1.7 = B2.0.

А после прошивки в zigbee версию можно HW как то посмотреть? Где берем - работали) у меня несколько 20 года датчиков - там вроде B1.4, несколько B1.6

devbis, 2023-10-28:
Версия платы написана на самой плате, со стороны экрана

zzz676, 2023-10-28:
> Где вы берете такие старые термометры - Hardware Revision String: B1.5 (производились более двух лет назад)? B1.5 наверно имеет самое большое потребление из всех из-за своего набора микросхем. [...]

Кстати Сегодня пришли 2 новых термометра. Оба версии 1.5. Дата производства 10.2022 .
[image]

pvvx, 2023-10-28:
> Кстати Сегодня пришли 2 новых термометра. Оба версии 1.5. Дата производства 10.2022 .

Это скорее всего дата упаковки. Сенсор SHTC3 уже давно не производится. Чип, заменяющий контроллер LCD в B1.5 и B1.6, устанавливался какой-то временный (похоже на како-то MCU со своей программой и управляемый по UART),  т.к. было время "дефицита чипов".

IvanGavrilenko, 2023-11-06:
> Версия платы написана на самой плате, со стороны экрана

Подскажите прошивал через UART и похоже оторвал резиновый шлейф от дисплея - теперь ничего не показывается (а может криво прошилось и не стартует плата) - вопрос не сталкивались ли с таким и как можно локализовать проблему (например померить на контрольных точках) ?

devbis, 2023-11-06:
Резинка не приклеена, просто прижимается платой и корпусом к экрану. С ней ничего не случится

Переполюсовка могла повредить чип, я не проверял:)

Если дело в прошивке, надо прошить еще раз на более низкой скорости или дополнительно стереть флешку

IvanGavrilenko, 2023-11-06:
> Переполюсовка могла повредить чип, я не проверял:)

А есть вариант проверить жива ли плата простыми действиями (вроде если дать на плату 3v3 то между gnd и 14 точкой должно быть 3v3) или нет и проще сохранить корпус этого термометра как памятник кривым ручкам?
 
>  или дополнительно стереть флешку

Я правильно понял что для этого запустить скрипт с ключом 'ea' вместо 'wf'?

devbis, 2023-11-07:
Я, к сожалению не знаю как можно проверить живость чипа, кроме как прошить и получить результат.

Да, еа стирает флешку

psike01, 2023-11-08:
Попробовал седня прошить 2 датчика с Hardware Revision String: B1.4 и Hardware Revision String: B1.6.
С 1.4 прошились без проблем, с 1.6 не хотят. Выходит ошибка
NotSupportedError: GATT operation failed for unknown reason.

Так что ошибка явно связана с ревизией датчика, а не действиями пользователя

devbis, 2023-11-08:
Я тестировал только на 1.4, у меня она в наличии. Там отличие только в экране должно быть. Интересно, поможет ли в этих случаях прошивка с выключенным экраном.

booroondook-eng, 2023-11-08:
Я несколько раз и прошивал с выключенным экраном, и спаривал с выключенным экраном. Более того - даже без батарейки (питание 3.3 В шло от USB-адаптера). Все работало.

pvvx, 2023-11-09:
> Так что ошибка явно связана с ревизией датчика, а не действиями пользователя

Ошибка связана с промежуточной прошивкой от devbis без исходников и уточнить что там происходит нет никаких возможностей. Только гадать.

devbis, 2023-11-09:
Прошивка совпадает с pvvx/atc_mithermometer 4.4 с добавлением перемещения прошивки по адресу 0x40000 и изменением 
`bls_ota_set_fwSize_and_fwBootAddr(200, 0x40000);`
при старте

Вот полный патч, но это первая рабочая версия, и я на ней остановился.
[ota_40000.patch](https://github.com/devbis/z03mmc/files/13305709/ota_40000.patch)

pvvx, 2023-11-09:
Следующее гадание, на основе лога:
```
20:42:53: Update error: NotSupportedError: GATT operation failed for unknown reason.
20:42:53: Disconnected.
```
говорит о том, что происходит разрыв связи. Причина может быть в элементарном нарушении таймингов BLE.
Скорее всего в прошивку вписана какая-то процедура, которая выполняется более нескольких мкс не позаботившись о таймингах BLE, например стирание сектора или секторов Flash....

pvvx, 2023-11-09:
Область Flash, куда записывается OTA для Zigbee 0x40000+ используется для сохранения логов замеров (названа memo).
Функция OTA в SDK не очищает область на ходу, куда записывается новый OTA. 
Очистка области для OTA производится только при старте, да с частичной проверкой на "пустоту" сектора путем проверки первых байт на 0xff.

kachalovmg, 2023-11-10:
[pvvx](https://github.com/pvvx), спасибо за альтернативный прошивальщик #50 
Прошил с помощью него датчик, который оставался новым. Первые два, с описанной здесь проблемой, прошитые промежуточной, не получилось. 
Есть ли возможность откатиться с промежуточной, что бы в дальнейшем прошиться, альтернативным прошивальщиком. Не знаю как правильно это сделать.

pvvx, 2023-11-10:
> Не знаю как правильно это сделать.

Если в датчик уже записана [ATC_ota_400000](https://github.com/devbis/z03mmc/blob/master/assets/ATC_ota_40000.bin), тогда только путем аппаратной перепрошивки (проводами).

Можно попробовать, если ещё работает ATC_ota_400000, перед самой перепошивкой нажать 
[image]. Но уверенности, что это  поможет - нет. Я не использовал и не тестировал ATC_ota_400000 от devbis.

kachalovmg, 2023-11-10:
[pvvx](https://github.com/pvvx)
Да, ATC_ota_400000 работает, но !Delete all records! не помогло, заказал USB/ttl. 
Спасибо.

pvvx, 2023-11-10:
@kachalovmg - Для выяснения ситуации есть просьба:
Запустите [DebugFlashRead.html](https://pvvx.github.io/ATC_MiThermometer/DebugFlashRead.html).
Соединитесь с термометром, впишите `40000`, затем `Read`, потом скопируйте сюда, что выйдет в последней строке лога:
[image]

kachalovmg, 2023-11-11:
> в последней строке лога:

040000:2680000000005D024B4E4C54270788008680

pvvx, 2023-11-11:
> 040000:2680000000005D024B4E4C54270788008680
Значит там сама прошивка сидит.
Тогда нужно смотреть ещё адрес 0

kachalovmg, 2023-11-12:
> Тогда нужно смотреть ещё адрес 0 и 0x20000

Адрес 0
`000000:FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF`
Адрес 0х2000
`002000:FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF`
Это означает пусто, я полагаю.

pvvx, 2023-11-12:
> Это означает пусто, я полагаю.

Да.  Это значит, что вероятнее всего просто сбой связи из-за срыва таймингов BLE какой-то длинной процедурой.

IvanGavrilenko, 2023-11-15:
> кроме как прошить и получить результат

Да действительно после очистки с помощью `python TLSR825xComFlasher.py -p COM9 -t5000 ea` и заново прошивки ошибка ушла и датчик прошился нормально.

Огромное спасибо за помощь

devbis, 2023-12-14:
I received V1.6 HW sensors and confirm that OTA is broken.
After some research I found out that enabling uart breaks the ability to use OTA on transitional firmware.
For this case I just disable LCD for 1.5/1.6 revisions in transitional fw, now update works correctly. 
For already stuck devices the only known way is to flash via UART/SWS.

closed 2023-12-14 by devbis