devbis/z03mmc #25: OTA Update

DimChik2903, 2023-10-26:
Is it possible to update the firmware via Zigbee2MQTT? If so, please add instructions.

I currently have version 1.0.2, but I would like to upgrade to 1.0.3

devbis, 2023-10-26:
You'll receive update once PR https://github.com/Koenkk/zigbee-OTA/pull/378 is merged.

If you need that urgently, you can set up local OTA index following this instruction https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

If you use ZHA for managing devices, you could follow this instruction https://community.home-assistant.io/t/zha-ota-firmware-update-upgrade/488960 (or search for a newer/more descriptive)

DimChik2903, 2023-10-26:
Как всё-таки лучше тут общаться, на русском, или на английском?)

Про Local OTA index я видел, но не понял, что в этом файле писать. Есть возможность добавить этот индекс в релиз тут, на гитхабе?

Mostalk, 2023-10-27:
@DimChik2903, You need to wait until the changes in the OTA index are updated, as I understand, this usually happens within a day.

But if you want to update it earlier, you need to create an index.json file in the zigbee2mqtt folder and paste the following there
```

[
    {
        "fileVersion": 268644353,
        "fileSize": 129362,
        "manufacturerCode": 4417,
        "imageType": 515,
        "sha512": "705a2c81f075686b6cf4880a4c04d10f641d65f5d3bb8cfa849579b206391c72ca3ed72e0610fc1f0855e069f9135e70a8f7d1d2490037c3bbf26fc0f8076632",
        "url": "https://github.com/devbis/z03mmc/releases/download/1.0.3/1141-0203-10033001-z03mmc.zigbee",
        "path": "images/Telink/1141-0203-10033001-z03mmc.zigbee"
    }
]

```
**NOTE: This code will only work for updating to version 1.0.3**

I did this and everything was updated

devbis, 2023-10-27:
Note, that it is a newer version 1.0.4 with fixed issue. Please use it.

DimChik2903, 2023-10-27:
Thanks, it worked!

closed 2023-10-27 by DimChik2903

adorobis, 2023-10-31:
> Note, that it is a newer version 1.0.4 with fixed issue. Please use it.
> 
> ```json
> [
>      {
> 
>          "fileVersion": 268709889,
>          "fileSize": 129378,
>          "manufacturerCode": 4417,
>          "imageType": 515,
>          "sha512": "54c954123b43a4c41376f57e4ab1869dca2698c178a82251725174e976da9107ebb7ddf632aa8223ba49a6b5e3dba7a878db7dcf801dcc5c3b888c2d5c1378a5",
>          "url": "https://github.com/devbis/zigbee-OTA/raw/master/images/Telink/1141-0203-10043001-z03mmc.zigbee"
>      }
> ]
> ```

What would be the right info for 1.0.5?

Mostalk, 2023-10-31:
> What would be the right info for 1.0.5?

I used this code

```
[
     {
         "fileVersion": 268775425,
         "manufacturerCode": 4417,
         "imageType": 515,
         "url": "https://github.com/devbis/z03mmc/releases/download/1.0.5/1141-0203-10053001-z03mmc.zigbee"
     }
]
```

adorobis, 2023-10-31:
> I used this code

Thanks, works great!

DimChik2903, 2024-02-24:
Does anyone have the contents of the file to update to the latest firmware?

Or should updates come automatically without ota override?

devbis, 2024-02-24:
If you use zigbee2mqtt and installed firmware from this repo, z2m offers you the latest firmware

DimChik2903, 2024-02-25:
> If you use zigbee2mqtt and installed firmware from this repo, z2m offers you the latest firmware

I have the firmware from this repository, but clicking on the “check for updates” button results in the following error:
`2024-02-25 15:49:37Failed to check if update available for 'Winter Garden Temp/Hum Sensor' (OTA: Device didn't respond to OTA request)`

[image]

DimChik2903, 2024-03-07:
The problem was solved by updating the coordinator firmware