devbis/z03mmc #26: Humidity sensor not working   Firmware version  1.0.3

KleinerSource, 2023-10-27:
[image]
The LCD screen displays a humidity of 0

Mostalk, 2023-10-27:
Сonfirm. I have two sensors. Both updated to 1.0.3.
One is fine, the other also has a problem with humidity
[image]

SergeyKa2021, 2023-10-27:
The same problem. HW B1.4.

devbis, 2023-10-27:
Please, check version 1.0.4, readings conversion is fixed

Mostalk, 2023-10-27:
Thanks. Now everything works!

```
{
    "battery": 74.5,
    "humidity": 59.7,
    "last_seen": "2023-10-27T08:37:48.151Z",
    "linkquality": 112,
    "temperature": 20.37,
    "update": {
        "installed_version": 268709889,
        "latest_version": 268709889,
        "state": "idle"
    },
    "voltage": 2700,
    "update_available": null
}
```

closed 2023-10-27 by devbis