devbis/z03mmc #28: Broken after upload firmware

Semyonchick, 2023-10-28:
Not working after upload z03mmc.bin
Flashing complete sucessfull, but after device not work

devbis, 2023-10-28:
Please, try to erase flash and write it again. You could lower the baudrate while writing

Timmiej93, 2023-10-28:
I think this may be a cause of wirelessly flashing this firmware directly without flashing the [BT firmware in step 4](https://github.com/devbis/z03mmc#flashing-from-a-custom-bluetooth-firmware-easy-way-for-new-devices) first. I did the same to one of my devices by accident, and it indeed doesn't do anything anymore after that.

Semyonchick, 2023-10-28:
@Timmiej93 before with I wright ATC_Thermometer.bin 
You restore it?

Timmiej93, 2023-10-28:
> You restore it?

No, I was ordering a few new ones anyway since I needed more, so I just ordered one more. The time and effort to figure out what was going on, how to flash it wired and probably even get the hardware wasn't worth it.

> before with I wright ATC_Thermometer.bin

I'm afraid that's not the right firmware. You should really follow the instructions I linked closely, that has worked for me every time. Don't skip any steps. Just so we're clear, these are the steps:

## Flashing from a custom bluetooth firmware (easy way for new devices)
1. Open an awesome tool from ATC_MiThermometer https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
2. Click "Connect" button and find device LYWSD03MMC, wait connection (Connected in logs)
3. On new device with stock firmware click "Do Activation" and wait some time
4. Next "Select Firmware", choose file with custom bluetooth firmware [ATC_ota_400000](./assets/ATC_ota_40000.bin), click "Start Flashing"
5. You will see in logs "Update done after NN seconds"
6. Reboot device removing battery, refresh web-page with flashing tool and connect device again (with name ATC_802190 or similar, based on mac-address). Do not change any control on page
8. Flash latest [z03mmc.bin](https://github.com/devbis/z03mmc/releases) firmware over transitional firmware to convert it to zigbee. Use https://devbis.github.io/telink-zigbee/ page if previous flasher stops because of the firmware size.

([source](https://github.com/devbis/z03mmc#flashing-from-a-custom-bluetooth-firmware-easy-way-for-new-devices))

Semyonchick, 2023-10-28:
Ok. Thanks.

EinSchwerd, 2023-10-30:
I'm having the same issue. I flashed the ATC firmware first, once it succeeded I flashed the z03mmc firmware with the following log:
`06:40:59: File: z03mmc.bin
06:40:59: File size: 129316 bytes
06:40:59: Count: 8083
06:41:01: Start DFU
06:42:28: Update done after 86.311 seconds
06:42:32: Disconnected.
06:54:15: Searching for devices
06:54:20: NotFoundError: User cancelled the requestDevice() chooser.`
When the flashing was complete the display went blank. I tried reinserting the battery but no luck.
Any suggestions on how to recover the unit? Or is it permanently bricked?

devbis, 2023-10-30:
You should use UART to write firmware in your case. Follow the steps in Readme in this repo.

enerbe, 2023-10-30:
I had the same blank screen issue, as i had ATC already on the device, i just uploaded the z03mmc.bin. Its my bad, i had to follow the instructions...
Tried to reinserted the battery and then shorted gnd and reset pins. Didn't help. So i connected to my Connect programmer(https://soldered.com/product/connect-programmer/) uart and tried to flash (gnd->gnd, twd->P14, 3v3->+) with :
python3 TLSR825xComFlasher.py -p /dev/tty.usbserial-1440 -t5000 wf 0 z03mmc.bin 
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/tty.usbserial-1440, 921600 baud...
Reset module (RTS low)...
Activate (5000 ms)...
Chip sleep? -> Use reset chip (RTS-RST): see option --tact

Did i miss something?

devbis, 2023-10-31:
Here is a soldering schema. 

[image]

The command looks okay. But you'd better follow the instruction in README as is covers some cases

adorobis, 2023-10-31:
I have such case as well. So I flashed with serial - connecting as on the above picture. Flashing finishes successfully but nothing is displayed on the screen neither the device is discoverable over bluetooth. Tried stock fw as well as some custom ones.

devbis, 2023-10-31:
Try to lower baudrate to 340000

athenard, 2023-11-05:
Hi, same issue here, successful flash via bluetooth (already had custom fw) But blank screen after flash. I just tried to flash it via serial, again successfully, but nothing more on the screen. @adorobis, did u find what the issue was ?

adorobis, 2023-11-06:
Not yet, suggestion above was to change the baudrate. Will try it in free time. Purchasing a new device was faster 😎

athenard, 2023-11-06:
Hey @adorobis I found the issue. I first had to flash via TTL the bluetooth custom firmware to make it work again. I then flashed again the zigbee firmware and it worked like a charm :)

adorobis, 2023-11-06:
Great, which Bluetooth firmware have you flashed via TTL? I've tried a few but none has worked...

athenard, 2023-11-06:
I flashed this one 
https://github.localhost/assets/ATC_ota_40000.bin

mribeiro, 2023-11-07:
> Here is a soldering schema.
> 
> [image]
> 
> The command looks okay. But you'd better follow the instruction in README as is covers some cases

May I piggyback on this post?

I used wago connectors instead of soldering wires (well... I tried first but the error is the same though) and here's how I did it (my soldering skills are crap but I think you can notice it):

[image]

Do you see anything wrong in this wiring assuming the connectors and cables are all OK?

The error I get is

```
❯ python3 TLSR825xComFlasher.py -p /dev/tty.usbserial-1130 wf 0 ATC_ota_40000.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/tty.usbserial-1130, 921600 baud...
UART-SWS 92160 baud. SW-CLK ~26.3 MHz(?)
Inputfile: ATC_ota_40000.bin
Write Flash data 0x00000000 to 0x00014254...
Error Read Flash Status! (0)
Error write sector at 0x000300!
```

The sector may vary between 003, 002, 006, etc...

Thanks

mribeiro, 2023-11-07:
> @mribeiro i had the same issue on mac. Changing timeout to 0.05 or 0.1 helped me.
> 
> https://github.com/devbis/z03mmc/blob/ec3f46c750dd31bb2b62b25b80c866a6ad4a0255/TLSR825xComFlasher.py#L484C2-L484C2
> 
> Play with the value, it may help you too

it worked with 0.5! awesome! many thanks

pvvx, 2023-11-08:
TLSR825xComFlasher.py - Does not work on USB-COM adapters that have LEDs on at the RX input.

devbis, 2023-11-08:
It works with USB-COM adapters with LEDs on RX/TX or whatever. Maybe it is not officially supported.

skyflyer, 2023-11-30:
> TLSR825xComFlasher.py :
> 
> * Does not work on USB-COM adapters that have LEDs on at the RX input.
> * Does not work on USB-COM adapters that have FTDI chip

Why doesn't it work with FTDI? Can you recommend a USB-TTL adapter (one that works with Mac) that works?

I flashed it over bluetooth (following the steps); the display works, but I can't connect it to either bluetooth or zigbee since.

Thanks!

pvvx, 2023-12-01:
* The FTDI chip receiver discards incorrectly timed symbols from the buffer and sets the receive error flag.
* The current through the resistor is not enough to light the LED and the levels “0” and “1” are shifted.

skyflyer, 2023-12-01:
Thanks for the explanation. But which one would you recommend? Any one without the FTDI chip?

pvvx, 2023-12-01:
None of the alternative firmware creators work on MAC. Or I don't know about it - I haven't seen it.
MAC is made for a different job.

skyflyer, 2023-12-01:
Ok 😊
But which usb serial device would you recommend that works for this purpose (win or linux then)

pvvx, 2023-12-01:
Any Chinese USB-UART chip. There are no such strict checks for full compliance of the bit diagram with the UART protocol. There are no 3 checks of each bit at calculated intervals. :)

marekorok, 2024-01-25:
> 340000

well, this tip actually this worked for me, almost...
I was struggling for several hours, always had success with default 921600 but device never worked, tried multiple times and on two devices, same result. Strange isn't?

with 340000 no luck, no communication:
```
 python3 TLSR825xComFlasher.py -p COM18 -b340000 -t2000 wf 0 ..\devbis_z03mmc_106.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open COM18, 340000 baud...
Reset module (RTS low)...
Activate (2000 ms)...
Low UART baud rate!
Chip sleep? -> Use reset chip (RTS-RST): see option --tact
```

**BUT**, with 460000 success flash and working device! both!
```
python3 TLSR825xComFlasher.py -p COM18 -b460000 -t2000 wf 0 ..\devbis_z03mmc_106.bin 
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open COM18, 460000 baud...
Reset module (RTS low)...
Activate (2000 ms)...
UART-SWS 46000 baud. SW-CLK ~23.9 MHz(?)
Inputfile: ..\devbis_z03mmc_106.bin
Write Flash data 0x00000000 to 0x0001fbd4...
-------------------------------------------------------
Worked Time: 85.938 sec
Done!

```
Thank you!
Two devices unbricked :-)

exen904, 2024-02-03:
I have a similar issue here, with a UART thats likely the same that @mribeiro used. At least they look the same. Connected like in the schematic with the resistor. As flasher, I used the [Web-Tool](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html). According to the log, erase flash and flash of the FW work fine, but after that, the display stays dead and I dont have any pairing information in Z2M. 

[image]

Im not sure if the issue is that I used the web flash writer? Also played with the baudrate and activation time a bit with no other results.

pvvx, 2024-02-03:
[image]
5000 ms

1. Turn off the power to the thermometer.
2. Disconnect the TX wire from the SWS
3. Run "Write to Flash"
4. While "Atime" is working, connect TX to SWS, then 3.3V to +Vbat.

exen904, 2024-02-04:
```
12:36:05: USB-COM opened.
12:36:08: File was selected, size: 130004 bytes
12:37:53: Reset DTR/RTS (100 ms)
12:37:53: Activate (3 sec)...
12:37:56: Write 130004 bytes in to Flash...
12:38:40: Done (46.936 sec).
12:38:40: Soft Reset MCU
12:40:28: Reset DTR/RTS (100 ms)
12:40:28: Activate (8 sec)...
12:40:36: Write 130004 bytes in to Flash...
12:41:19: Done (51.902 sec).
12:41:19: Soft Reset MCU
```

Well, at least the runtime was longer this time. Though, still no display output. 

Connections as follows:

[image]

Starting with only RX and GND connected to the thermometer. The TX and 3.3V connection made on the breadboard only as soon the flasher is at "atime"

lbarjak, 2024-02-04:
Worked for me:
python3 TLSR825xComFlasher.py -p /dev/ttyUSB0 wf 0 z03mmc.bin
https://blogger.googleusercontent.com/img/b/R29vZ2xl/AVvXsEiZ898MZYEOIkzdRSRhFU6ojQcwlmOnxzt-qnAVhDqMgYTsttDPgoE9pvYVnQrO6upH8maowqxbIrlgi8tkv49amD-fhsAhFfU6VMOImIvnte3vDZqA8uIZprfY0wPgaJEr49Gi61Ur2AFEyKysRM4tsU0jTCdrP6M7cjdLW6cpX7OMFdpGZdE-77yD8l-x/s4000/IMG_20231207_234335.jpg

Daniel Gotteswinter ***@***.***> ezt írta (időpont: 2024.
febr. 4., V, 13:10):

> 12:36:05: USB-COM opened.
> 12:36:08: File was selected, size: 130004 bytes
> 12:37:53: Reset DTR/RTS (100 ms)
> 12:37:53: Activate (3 sec)...
> 12:37:56: Write 130004 bytes in to Flash...
> 12:38:40: Done (46.936 sec).
> 12:38:40: Soft Reset MCU
> 12:40:28: Reset DTR/RTS (100 ms)
> 12:40:28: Activate (8 sec)...
> 12:40:36: Write 130004 bytes in to Flash...
> 12:41:19: Done (51.902 sec).
> 12:41:19: Soft Reset MCU
>
> Well, at least the runtime was longer this time. Though, still no display
> output.
>
> Connections as follows:
>
> image.png (view on web)
>
> Starting with only RX and GND connected to the thermometer. The TX and
> 3.3V connection made on the breadboard only as soon the flasher is at
> "atime"
>
> <https://github.com/devbis/z03mmc/issues/28#issuecomment-1925727386>, or
> unsubscribe

michal-demecki, 2024-02-21:
ch340g did not work for me. 
cp2102 worked but flushing did not solve zigbee pairing issue. 
Flushing back to bluetooth solved connectivity problem, but flushing through OTA "bricked" it again (shows temperature, but can't connect to HA).
Zigbee works fine for 11 other devices (4 older, 7 newer from same batch).
Any suggestions how to debug/fix it?

devbis, 2024-02-22:
The fix workflow that I use to recover dead sensors:
1. Use python script only, web flasher doesn't work for me
2. Use `ea` command to clear all flash
3. Write (`wf`) final  firmware, zigbee or ble, depends on what you need. If didn't work, start from (2)

Sometimes playing with timeouts in the script works, but it is not like something standard for all cases. I use timings that are in this repo, or you may try ones from pvvx's repositories.

tovy14, 2024-03-15:
Sorry to piggyback of this thread but here it goes:
After flashing the 1st sensor, i thought i would do them all, boy i was wrong. I managed to rip the P14 pad off the board while soldering.

Does someone have a schematic for where the P14 pad is connected to on the board.

Thank you.

devbis, 2024-03-15:
@tovy14  It goes directly to the tlsr SoC, you can use a pinout image from the datasheet for 24-pin chip http://wiki.telink-semi.cn/doc/ds/DS_TLSR8251-E_Datasheet%20for%20Telink%20BLE+IEEE802.15.4%20Multi-Standard%20Wireless%20SoC%20TLSR8251.pdf , the pin number is 4.

nmaupu, 2025-02-01:
I confirm that CH340G is working perfectly for me and steadily with default commands (see readme)
I tested CP2102 and FTDI without success.
Maybe, it should be a good idea to add this information in the readme as well as be more precise about the SWS needs which could be counter intuitive when you are not that familiar with the serial protocol.