devbis/z03mmc #29: cant connect after flash  firmware ATC_ota_400000, hw B1.7

freeskyflying, 2023-10-28:
everything is ok from step 1-4
but after step 4 , flash  firmware ATC_ota_400000
when i reboot device ,reflash website
connect shows:
21:50:33: Reconnect 5 from 5
21:50:33: Hardware Revision String: B1.7
21:50:33: Software Revision String: V4.4
21:50:33: Firmware Revision String: github.com/pvvx
21:50:33: Detected Mi device
21:50:33: NotSupportedError: GATT operation failed for unknown reason.
21:50:33: Something went wrong, too many reconnects

Mostalk, 2023-10-28:
I had a similar error when I flashed from a computer, try from your phone

booroondook-eng, 2023-10-30:
Try to extract and insert the battery.

adamlc, 2023-10-31:
Also try a different battery. I've found if the battery is a bit on the low side it may throw random errors like above

MartynIden, 2023-10-31:
i  have a problem with hw B1.7 revision too, brick after first flashing via air (no connect, no screen), then flashed via UART - all worked fine but get another brick after OTA (z2m) on ver 1.0.3
reflashed 1.0.3 via UART again and brick after OTA to 1.0.5

revision 1.7 little shitty)) another device hw B1.4 worked fine

Adrian-czw, 2023-10-31:
I have exactly the same symptom hw B1.7 - after uploading the software and rebooting, the sensor screen displays nothing. Battery replaced with new one and reset done.

With another hw B1.4 device, the transition to Zigbee was successful.

devbis, 2023-10-31:
I don't have any idea why OTA on B1.7 works incorrectly. Feel free to share your thoughts.

pvvx, 2023-11-02:
For B1.7, B1.9, B2.0, the code from ATC1441 used to work with the display controller in the Zigbee firmware is not quite correct.
Failures often occur and the display readings fly off. 
Partially helps to reset the power supply...

mtdn, 2023-11-07:
Same error with B1.6. Tryed on several devices with PC and phone. 

Flashing ATC_ota_40000 is OK. After reconnect:
```
Status: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.4
Vbat: 3113 mV , Temp: 25.97°C, Humi: 52.32%, ID: 58, Reed switch counter: 0, flg: 0x04:r0/t0
```

```
14:06:50: Searching for devices
14:06:56: Connecting to: ATC_203A9D
14:07:00: Hardware Revision String: B1.6
14:07:00: Software Revision String: V4.4
14:07:00: Firmware Revision String: github.com/pvvx
14:07:00: Detected custom Firmware
14:07:00: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.4, Sensor: SHT4x
14:07:00: Custom config HEX string: 55051000002804a9313104b4
```

And when trying to flash anithing after ATC_ota_40000.bin (not only z03mmc.bin) I have error "Update error: NotSupportedError: GATT operation failed for unknown reason."

### I think that is important: all firmwares stops on "Sending block nr: 7 from ..." when flashing
```
Status: Sending block nr: 7 from 8078, 0% done, time since start 0.02s
Vbat: 3113 mV , Temp: 25.97°C, Humi: 52.32%, ID: 58, Reed switch counter: 0, flg: 0x04:r0/t0
```

```
14:06:50: Searching for devices
14:06:56: Connecting to: ATC_203A9D
14:07:00: Hardware Revision String: B1.6
14:07:00: Software Revision String: V4.4
14:07:00: Firmware Revision String: github.com/pvvx
14:07:00: Detected custom Firmware
14:07:00: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.4, Sensor: SHT4x
14:07:00: Custom config HEX string: 55051000002804a9313104b4
14:14:31: File: z03mmc.bin
14:14:31: File size: 129236 bytes
14:14:31: Count: 8078
14:14:34: Start DFU
14:14:40: Update error: NotSupportedError: GATT operation failed for unknown reason.
14:14:40: Disconnected.
```

pvvx, 2023-11-07:
@mtdn  - Clarification is not possible - there is no source code ATC_ota_40000.bin.

jumakki, 2023-11-11:
@mtdn, I had same issue with B1.6. Flashing with Android phone, same as I used for flashing ATC_ota_40000, failed repeatedly on block nr 7. But I was able to flash Zigbee successfully using this flasher https://github.com/devbis/z03mmc/issues/50 when using Linux PC.

nicolashimmelmann, 2023-11-20:
Had the same issue with revision B1.6 several times, on all 4 devices that I bought. Flashing ATC_ota_40000.bin went smoothly after activation, but flashing z03mmc.bin always gave an error after the first battery-out/in and connecting to the ATC_XXXX device.

What helped was simply retrying:
- just remove the battery, put it back in, reload the TelinkMiFlasher website and try to connect and flash z03mmc.bin again.
- If it still does not work, connect the GND and Reset Pins on the back side until the device shows up for connection, connect again on the flasher website, and flash z03mmc.bin again.
- just retry until it works.

I was able to flash all 4 devices after a maximum of 3 tries per device. But I have no idea what the cause for the issue is. As others reported above, it stops at "Sending block nr: 7 from ..." and shows the " NotSupportedError: GATT operation failed for unknown reason."

pvvx, 2023-12-03:
**z03mmc.bin** from **devbis** is not supported in BLE OTA [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).
**z03mmc.bin** does not support BLE OTA booting.
The use of `**ATC_ota_40000.bin**` from **devbis** is not recommended due to many problems for users.

[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) supports downloading Zigbee firmware in *.zigbee file format.

devbis, 2023-12-14:
I received V1.6 HW sensors and confirm that OTA in transitional firmware is broken.
After some research I found out that enabling uart breaks the ability to use OTA.
For this case I just disable LCD for 1.5/1.6 revisions in transitional fw, now update works correctly.
For already stuck devices the only known way is to flash via UART/SWS.

closed 2023-12-14 by devbis