devbis/z03mmc #30: Cant connect with teling flasher after update to zigbee fw

komw, 2023-10-30:
Its possible to flash overt OTA a flashed device with zigbee fw?
Currently my devices are connected via zigbee, but I'm unable to connect to it with the telink flasher. 
How to upgrade a fm when device uses a zigbee network?

devbis, 2023-10-30:
Telink OTA flasher uses bluetooth protocol while the sensor is converted to zigbee. It won't work.

Use zigbee tools to upgrade firmware:
1. Zigbee2mqtt: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files 
2. ZHA: https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

Other coordinators may not support OTA. In this case you should update via UART.

komw, 2023-10-30:
O, thx!
I not supposed that z2m support OTA from local. thx!

closed 2023-10-30 by devbis