devbis/z03mmc #31: Version: 2.1.1_0159 is temporarily not supported (!)

klu16, 2023-10-30:
Hello,

is there a plan to support this firmware version? If so, do you have an approximate date?

Thanks!

devbis, 2023-10-30:
It may require additional work from https://github.com/pvvx/ATC_MiThermometer/ project as the current installation heavily relies on the work done by pvvx.  You may track https://github.com/pvvx/ATC_MiThermometer/issues/378 for updates.
Additionally, you can solder wires to flash firmware using UART, current fw version won't affect this method.

VonalOrdu, 2023-11-02:
To go back to the old version (Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin );
To do this, you must remove your temperature sensor and connect it with the cables by following the steps below.
This is how I solved my problem.
I'm sorry for my bad english.

1. https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin download
2. https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html 
3. [image]
4. [image]
5. File select Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin
6. Write to flash

devbis, 2024-02-13:
Please, track https://github.com/atc1441/ATC_MiThermometer/issues/298 for updates

closed 2024-02-13 by devbis

spencermccarthy, 2024-03-26:
Is it possible to use the Xiaomi BLE integration on this new firmware? I guess not is a BindKey cannot be set.

devbis, 2024-03-26:
This firmare is to use the sensor with Zigbee network. For bluetooth support you'd better ask in BLE atc1441/pvvx repos

jksmurf, 2024-10-08:
> To go back to the old version (Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin ); To do this, you must remove your temperature sensor and connect it with the cables by following the steps below. This is how I solved my problem. I'm sorry for my bad english.
> 
> 1. https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin download
> 2. https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
> 3. 
> 5. File select Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin
> 6. Write to flash

I tried this with a FT232 TTL-USB but keep getting the message "_Select file is no telink firmware .bin_" and the "_Write to Flash_" button is always greyed out. I do not get the error message with other bins e.g. Z03MMC_v0122 but my H&T does not work even after it succesfully flashes. 

I orignally had 0159 FW but now the thing seems to be bricked. This and anotehr site that pointed ot this, suggested I could upload the old FW but I am struggling to do this.

I have GND on the FT232 to the Battery Terminal -ve (Centre of PCB), VCC to Vbat+ (Battery Edge Spring) and TX to Pin14. I also tried conecting the DTR to the Battery Terminal =ve (Battery Edge Spring [as suggested by pvvx.]([url](https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-1845242913))

Any ideas would be welcomed thank you!

[image]
[image]
[image]
[image]
[image]

martinlbb, 2024-10-08:
Got 5 devices with 2.1.1_0159 firmware, that are incompatible with OOb OTA update.

Did what was shown on this issue.
RS232 adapter on 3.3V
Only 4 wires:
* VCC (to battery holder, the outer one)
* GND (to battery holder, the center one)
* TX from RS232 to reset pad
* DTR to P14 pad (SWS pin).

Was using the USB COM flash page with Chrome and firmware 1.0.0_0130.

[image]

All 5 devices converted to older firmware, and then concerted to custom one.

Perfect!

jksmurf, 2024-10-09:
> Got 5 devices with 2.1.1_0159 firmware, that are incompatible with OOb OTA update.

Wll done to you! You have me hope, but alas, I must have tried 100 times, I cannot revive my device.

Despite [this site](https://smarthomescene.com/guides/convert-qingping-cgdk2-from-bluetooth-to-zigbee/) saying (generically) that you cannot brick these chips, I seem to have done that OR my connections are just not taking.

```
You cannot brick Telink TLSR82xx chips
No matter what you flash, you can always use UART to restore
```

Quick Q; DTR to reset, you mean the RESET PIN not the the GND Pin (also used ot reset)?

[image]
[image]
[image]
[image]

devbis, 2024-10-09:
@jksmurf You'd better use a different  UART dongle on a different chip.
And the python-based uart flasher along with resistor between rx-tx works better. At least you can be sure that the device is in the flashing mode and not just ignores packets from uart.

lbarjak, 2024-10-09:
https://barjak.eu/pictures/ft232rl-m.jpg
It worked for me.

jksmurf ***@***.***> ezt írta (időpont: 2024. okt. 9., Sze,
5:49):

> Got 5 devices with 2.1.1_0159 firmware, that are incompatible with OOb OTA
> update.
>
> Wll done to you! You have me hope, but alas, I must have tried 100 times,
> I cannot revive my device.
>
> Despite this site
> <https://smarthomescene.com/guides/convert-qingping-cgdk2-from-bluetooth-to-zigbee/>
> saying (generically) that you cannot brick these chips, I seem to have done
> that OR my connections are just not taking.
>
> You cannot brick Telink TLSR82xx chips
> No matter what you flash, you can always use UART to restore
>
> Quick Q; DTR to reset, you mean the RESET PIN not the the GND Pin (also
> used to reset)?
>
> TSLR82xx.Cannot.Brick.jpg (view on web)
> Trials.jpg (view on web)
> ThisSetup.jpg (view on web)
> Pinouts2.jpg (view on web)
>
> <https://github.com/devbis/z03mmc/issues/31#issuecomment-2401212566>, or
> unsubscribe

jksmurf, 2024-10-09:
> https://barjak.eu/pictures/ft232rl-m.jpg It worked for me.

Hmm... 

@Ibarjak I am now even more confused. 

Your pic (thank you) shows wiring for a FT232RL chip with what appears to be the1k resistor wiring from https://github.com/pvvx/TlsrComSwireWriter which says "_Does **not** work on USB-COM adapters that have FTDI chip_"

devbis (above) says use a DIFFERENT USB-COM Adapter, i.e. non -FTI (presumably) which matches what pvxx shows.

lbarjak, 2024-10-09:
https://www.hestore.hu/prod_10036483.html
Cheap copy, probably not original chip

jksmurf ***@***.***> ezt írta (időpont: 2024. okt. 9., Sze,
12:03):

> https://barjak.eu/pictures/ft232rl-m.jpg It worked for me.
>
> Hmm...
>
> @Ibarjak I am now even more confused.
>
> Your pic (thank you) shows wiring for a FT232RL chip with what appears to
> be the1k resistor wiring from https://github.com/pvvx/TlsrComSwireWriter
> which says "*Does not work on USB-COM adapters that have FTDI chip*"
>
> devbis (above) says use a DIFFERENT USB-COM Adapter, i.e. non -FTI
> (presumably) which matches what pvxx shows.
>
> <https://github.com/devbis/z03mmc/issues/31#issuecomment-2401885046>, or
> unsubscribe

jksmurf, 2024-10-11:
Just wanted to report I went and bought some 1kohm (up to 1.8kohm) resistors and a CH340G USB-TTL device with the **intention** to wire it up [like on here ](https://github.com/pvvx/TlsrComSwireWriter)and run the Python script ... 

... but then I thought, just for kicks, why not try the simple 4 wire setup **first** on this Chip (different to my red FTI FT232RL), but without the resistor and using the [Web-based TX-SWS Flasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html).

i.e. wiring FROM the Device TO the Board:

- GND to -3.3Vbat
- VCC to +3.3VBat
- TX to SW (Pin14)
- RTS to Reset (RST)

And of course, it worked FIRST time, with this model (I bought an old one, with no LEDs that I could see)., without resistors or Python scrips. So if you're not having much luck with a FTI FT232RL then maybe just try a CH340G, they are cheap, US$5. 
 
[image]

dodgydd, 2024-10-29:
> FT232 CH340G

Thanks, I'll google around. How do you keep the probes from the chip to the thermometer board steady? I'll be using dupont wires but 2 of the locations on the mijia board look quite hard to either solder (prefer not to solder) or to keep stead while it's flashed 😂

jksmurf, 2024-10-29:
> > FT232 CH340G
> 
> Thanks, I'll google around. How do you keep the probes from the chip to the thermometer board steady? I'll be using dupont wires but 2 of the locations on the mijia board look quite hard to either solder (prefer not to solder) or to keep stead while it's flashed 😂

Small light sprung crocodile clips on V+ and V-;
A very small hole in the housing you can poke a pin through to P14; an extra set of hands for the RST and computer :-) yup very very fiddly … but after many goes it becomes easier … I didn’t use the GND pin, just V- (bottom spring if battery), you can clip onto that. Good luck !

dodgydd, 2024-10-29:
> > I really appreciate your help, thanks! I've just got the dremel out and guesstimated first time where to drill a hole to expose p14, so now I have a convenient access and also some support for the probe! I'll return here with the exact board I bought and any updates to help others that follow. Thanks. !
> 
> I’d probably have made the hole quite a bit smaller as it holds the pin better but you’re on the right track. Thanks - pay it forward 👌

I've got 5 to do so hopefully the next 4 will have a smaller drill hole if I can find a tiny drill!

jksmurf, 2024-10-29:
> I've got 5 to do so hopefully the next 4 will have a smaller drill hole if I can find a tiny drill!

btw if you use these little spring loaded things, they always fall off the V+ and V- as the wires are too soft to grip properly; but I had better luck extending the wires (pressing the spring loaded end) and bending them inwards with a pair of tweezers so they hooked on better. YMMV.

[image]

jksmurf, 2024-10-30:
> I have a pack of 5 mijia thermometers I want to flash for home assistant (already have 5 attached no problem but they were bought 2 years ago!)

Just a comment on this statement as I wasn’t sure your reasons for wanting to get the USB-TTL device, but presumably you have bought devices with updated FW OR updated the FW by mistake OR had a failed update on one or more of those five devices, which is why you’re pursuing the UART route? 

Asking as just flashing using the [Telink Web based flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) should be the first thing you try provided the FW is OK and you don’t have to downgrade it (via UART).

See a couple of notes I wrote in the home-assistant forum.

[Note 1 re downgrading if in BLE](https://community.home-assistant.io/t/home-asssistant-xiaomi-mijia-lywsd03mmc-temperature-and-humidity-sensor-tutorial-september-2022/456403/149?u=smurfhk)

[Note 2 re downgrading if stuck in Zigbee mode when Telink no longer works](https://community.home-assistant.io/t/home-asssistant-xiaomi-mijia-lywsd03mmc-temperature-and-humidity-sensor-tutorial-september-2022/456403/152?u=smurfhk).

jksmurf, 2024-10-30:
> > FT232 CH340G
> 
> Thanks, I'll google around. How do you keep the probes from the chip to the thermometer board steady? I'll be using dupont wires but 2 of the locations on the mijia board look quite hard to either solder (prefer not to solder) or to keep stead while it's flashed 😂

I might have put you wrong with the CH340G note; I’ve amended my notes above to state that it says FT232 on the chip, BUT it was not supposed to be an FTI Chip, at least how it was marked in the shop. 

Looking at a few websites now though, it does [actually seem to have the FTDI chipset](https://instock.pk/usb-to-ttl-adapter-ftdi-usb-uart-ic-ft232rl-ft232.html)! So maybe it does work with any USB-TTL, that the change that made it work was the RTS to RST, I really don’t know. 

My FTI board (Red) didn’t have a RTS **pin** and I couldn’t be bothered trying to solder a wire on to the RTS connector hole (that you can actually see on that red board, about three holes down from the GRD pin).

dodgydd, 2024-10-30:
> > > FT232 CH340G
> > 
> > 
> > Thanks, I'll google around. How do you keep the probes from the chip to the thermometer board steady? I'll be using dupont wires but 2 of the locations on the mijia board look quite hard to either solder (prefer not to solder) or to keep stead while it's flashed 😂
> 
> Hi, sorry, I might actually have put you wrong with the CH340G note; I’ve amended my notes above to state that it says FT232 on the chip, BUT it was not supposed to be an FTI Chip, at least how it was marked in the shop.
> 
> Looking at a few websites now though, it does [actually seem to have the FTDI chipset](https://instock.pk/usb-to-ttl-adapter-ftdi-usb-uart-ic-ft232rl-ft232.html)! So maybe it does work with any USB-TTL, that the change that made it work was the RTS to RST, I really don’t know.
> 
> My FTI board (Red) didn’t have a RTS **pin** and I couldn’t be bothered trying to solder a wire on to the RTS connector hole (that you can actually see on that red board, about three holes down from the GRD pin).

Oops, I've already ordered some CH340g boards. Yes I have already tried to flash them OTA (I have done this before successfully with older mijia thermometers) but the firmware is incomptible and so needs to be rolled back via UART.

jksmurf, 2024-10-30:
> Oops, I've already ordered some CH340g boards. Yes I have already tried to flash them OTA (I have done this before successfully with older mijia thermometers) but the firmware is incomptible and so needs to be rolled back via UART.

Darn, my apologies for that you were more eagle eyed than me in your first post. I was just so ecstatic to revive it. If you’re ordering boards plural, can you maybe change the order so you order one CH340G and one FTDI FT232RL based one (with an RTS pin) instead?

dodgydd, 2024-10-30:
> Darn, my apologies for that you were more eagle eyed than me in your first post. I was just so ecstatic to revive it. If you’re ordering boards plural, can you maybe change the order so you order [...]

It's no big deal, these things are so cheap! I've also ordered  an FTDI FT232RL (it doesn't have an RTS pin, but there is a hole for RTS on the edge of the board I can connect to, if it can work that way!?)

jksmurf, 2024-10-30:
> It's no big deal, these things are so cheap! I've also ordered an FTDI FT232RL (it doesn't have an RTS pin, but there is a hole for RTS on the edge of the board I can connect to, if it can work that way!?)

Sounds good, I’m sure one of them will work for you. My apologies again, I had so many goes with the red one using the DTR connection pin (as advised elsewhere) I almost gave up, but persistence and sheer bl##dy mindedness made me try a different board with the RTS pin. 

Best of luck to you 👍

lits101, 2024-11-16:
I'm also getting a blank screen after trying the above method from @jksmurf, flashing direct to v4.8. Attempting to flash v1.0.0 yields the same result. I expect this is a result of a bad firmware write. 

[image]

I'm definitely using 3V3 on my FT232R board. 

[image]

I'm attempting this through the https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html tool, on macOS 15 and Chrome 130.0.6723.117

jksmurf, 2024-11-16:
Above I noted using this flasher: 

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

can you please give that a try instead?

k.

lits101, 2024-11-16:
> Above I noted I used this flasher:
> 
> https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
> 
> can you please give that a try instead?
> 
> k.

This is the Bluetooth OTA flasher, isn't it? I get the (!) Version: 2.1.1_0159 is temporarily not supported (!) error using that one.

jksmurf, 2024-11-16:
> > Above I noted I used this flasher:
> > https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
> > can you please give that a try instead?
> > k.
> 
> This is the Bluetooth OTA flasher, isn't it? I get the (!) Version: 2.1.1_0159 is temporarily not supported (!) error using that one.

lol 😂 my apologies yes, sorry, of course l, you’re trying to downgrade using USB UART, doh, my mistake.

Not sure, I just persisted with the flasher and the various connections.

jksmurf, 2024-11-16:
@lits101, Hi, bugs me that it doesn’t work for you, just reading your post again, presumably you downloaded the v1.0.0_0130.bin correctly, clicking the link to download it and and not right clicking, download?

And that you have RTS to RESET and all other probes on the correct pins.

My only other suggestion is order a different USB UART unit exactly like mine (I think there’s some AliExpress or Amazon li is above) and try that, they are quite cheap. 

According to the note I referenced above you cannot brick these Telink units so if you’re persistent, you might be successful. 

Best of luck!

piscopoc, 2024-12-03:
I seem to be in the same boat as @lits101 with the flasher, using a DSD Tech variant. Once I flashed the 130 bin I have a blank display now as well. 

Any advice on with USB UART has the most success?

jksmurf, 2024-12-04:
I am guessing you used one of these https://www.deshide.com/TTL-Series.html
TBH I am wary of giving advice as to which might work, I can only say which did for me, and the pin conenctions I used

i.e. as above, I [think this is my one on Amazon](https://www.amazon.com/Adapter-Serial-Converter-Development-Projects/dp/B075N82CDL) and on [AliExpress t](https://www.aliexpress.com/item/1005002740084525.html)

**Repeating the note above:**

Simple 4 wire setup using the [Web-based TX-SWS Flasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html).

Wiring FROM the Device TO the Board:

GND to -3.3Vbat
VCC to +3.3VBat
TX to SW (Pin14)
RTS to Reset (RST)

Let us know how you got on!

piscopoc, 2024-12-04:
@jksmurf thanks for the links! I will give that device a go, appreciate the support!

jksmurf, 2024-12-05:
Might be worth folks having a look at [this link too](https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-2465907616), a very good description:

See also the [Video here](https://www.youtube.com/watch?v=BtsxkOS6Zj8) where he describes which USB-TTL adapter he uses (in the comments).

FYI (only), in additin to mine above, I link the ones that used succesfully by others (and note some have issues with using them on a MAC):

- [USB2TTL_01](https://www.amazon.co.uk/dp/B09CK79B4C?ref=ppx_yo2ov_dt_b_fed_asin_title) : CH340C Chip
- [USB2TTL_02](https://aliexpress.com/item/1005002822094696.html) : FT232 FT232BL FT232RL
- [USB2TTL_02](https://www.amazon.de/dp/B08T24NML9?th=1) : CH340G Chip

ulich, 2024-12-15:
I also had quite some trouble flashing any firmware, it always ended with a blank screen. I finally managed to make it work, see
https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-2543904906

piscopoc, 2024-12-17:
Still no luck getting this to flash, not really sure what to make of this message either:

`Chip sleep? -> Use reset chip (RTS-RST): see option --tact`

Tired to just erase using the following command on both a Ubuntu and Mac but both return the same error with the new USB-TTL adapter that I purchased:

`python3 TLSR825xComFlasher.py -p /dev/ttyUSB0 -t10000 ea`

jksmurf, 2024-12-17:
> not really sure what to make of this message either:
> 
> `Chip sleep? -> Use reset chip (RTS-RST): see option --tact`

No idea I'm sorry, looking at others with that message ..seems like tact is activation time, but I've no idea what you need to do to get past it sorry :-( ... maybe get to the Pin P14 faster ... ? 

- https://github.com/pvvx/TlsrComSwireWriter/issues/1 
- https://www.reddit.com/r/hardwarehacking/comments/1g24glh/tlsr8250_howto_flash_it/

dodgydd, 2024-12-22:
Just popping back it to give an update. I got the USB flasher working with just 4 wires, no resistors as per @jksmurf directions. All flashed and updated no problems first time. I used https://www.aliexpress.com/item/1005002740084525.html which I believe is the same that jksmurf used, possibly different supplier.
Thanks for the continuing support youv'e shown to others on here @jksmurf and merry Christmas (if you celebrate it!).

jksmurf, 2024-12-22:
> Just popping back in to give an update. I got the USB flasher working with just 4 wires, no resistors as per @jksmurf directions. All flashed and updated no problems first time. I used https://www.aliexpress.com/item/1005002740084525.html which I believe is the same that jksmurf used, possibly different supplier. Thanks for the continuing support youv'e shown to others on here @jksmurf and merry Christmas (if you celebrate it!).

Great to hear that it worked for you and I really appreciate you coming back with your experience, good or bad, it might help others !

Merry Christmas to you too 🎄

mastersf, 2025-03-18:
why would the usb flasher hardware be important to this process ?

jksmurf, 2025-03-18:
> why would the usb flasher hardware be important to this process ?

No idea, technically. 
However if it works, flashers are cheap, so just use the one that works 👍🏼