devbis/z03mmc #32: How to do OTA?

kubert96, 2023-10-30:
I have aded and flashed device on 1.0.4 and now 1.0.5 is out. 
How to do ota? 
Bluetooth is not working as expected. 
And in zigbee2mqtt OTA doesnt register any update. 

installed_version: 268709889
latest_version: 268709889

 Update available: false

Thanks :)

Mostalk, 2023-10-31:
Use this in "OTA index override"

```
[
     {
         "fileVersion": 268775425,
         "manufacturerCode": 4417,
         "imageType": 515,
         "url": "https://github.com/devbis/z03mmc/releases/download/1.0.5/1141-0203-10053001-z03mmc.zigbee"
     }
]
```

kubert96, 2023-10-31:
error 2023-10-31 07:19:44Update of 'Office Aquara Temp' failed (Device didn't respond to OTA request)

kubert96, 2023-10-31:
info 2023-10-31 08:42:28MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Office Aquara Temp"},"error":"Update of 'Office Aquara Temp' failed (Device didn't respond to OTA request)","status":"error","transaction":"6594s-2"}'
error 2023-10-31 08:42:28Update of 'Office Aquara Temp' failed (Device didn't respond to OTA request)

Mostalk, 2023-10-31:
@kubert96 
Hmm, it worked for me. With such an error, it helped me to reset the battery.

In any case, you can just wait until @devbis updates the OTA index repository.

devbis, 2023-10-31:
https://github.com/Koenkk/zigbee-OTA/pull/386

PR to track

kubert96, 2023-10-31:
info 2023-10-31 09:20:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Office Aquara Temp"},"error":"Update of 'Office Aquara Temp' failed (Unexpected string in JSON at position 160056)","status":"error","transaction":"gbkbn-2"}'

New Error... :D

Nab0y, 2023-10-31:
One of my eight sensors also refuses to update via OTA, but it reports a slightly different error. 
`Info 2023-10-31 14:12:00MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Температура влажность кухня"},"error":"Update of 'Температура влажность кухня' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"j9w29-1"}'
Error 2023-10-31 14:12:00Update of 'Температура влажность кухня' failed (Timeout: device did not request any image blocks)`
I suspect that it's because this sensor is the only one currently connected not directly to the coordinator but to the nearest router, and the issue might not be with the sensor itself but with the router, which seems to be refusing to pass on the update through itself. Well, that's my theory, at least.
[image]

kubert96, 2023-10-31:
Mine is directly conected to Cordinator. 

[image]

devbis, 2023-10-31:
> device did not request any image blocks
usually means that OTA image and fw version is the same. But if it is not, you may replug the battery and push OTA again

Nab0y, 2023-10-31:
> Mine is directly conected to Cordinator.
> 
> [image]

Yes, it's not my case. You can try to forcefully make the sensor take the firmware from a local folder.
[image]
[image]
After that, you'll need to restart Z2M since the configuration file has been modified. And once you're done, don't forget to revert everything back to its original state.

Nab0y, 2023-10-31:
> > device did not request any image blocks
> 
> usually means that OTA image and fw version is the same. But if it is not, you may replug the battery and push OTA again

That's what I intend to do, but it will take me a few hours to physically reach the sensor :)

kubert96, 2023-10-31:
This is after i restarted sensor: 
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Office Aquara Temp"},"error":"Update of 'Office Aquara Temp' failed (localIndex.concat(...).map is not a function)","status":"error","transaction":"pfbx3-3"}'

kubert96, 2023-10-31:
OK :D 

info 2023-10-31 13:05:42MQTT publish: topic 'zigbee2mqtt/Office Aquara Temp', payload '{"battery":90,"humidity":62.36,"linkquality":102,"temperature":24.06,"update":{"installed_version":268709889,"latest_version":268775425,"progress":0,"state":"updating"},"update_available":false,"voltage":2900}'

Finaly working. With Local File. :) 

@Nab0y thank you :)

adorobis, 2023-10-31:
> usually means that OTA image and fw version is the same. But if it is not, you may replug the battery and push OTA again

For me some devices don't show correct fw version information in z2m (not in HA) although they were updated successfully. I have such case for 1 of 3 of the temp sensors but also some other devices (e.g. xiaomi wall switches). Re-inserting batteries and clicking "check for updates" does not help. Example below. Anyway it seems a z2m issue as it is not limited to those sensors.
[image]

devbis, 2023-10-31:
@adorobis You can read newer values for devices using dev console for a device
[image]

adorobis, 2023-10-31:
I get either timeout or nothing changes. Still shows the old version in z2m.

fraddygil, 2023-10-31:
Hey Guys , Is there any way to go back to the original firmware or any ATC firmware?

Nab0y, 2023-10-31:
> Hey Guys , Is there any way to go back to the original firmware or any ATC firmware? Using OTA or another option?

https://github.com/devbis/z03mmc#flashing-firmware-with-usb-to-uart

shirou93, 2023-11-01:
> Yes, it's not my case. You can try to forcefully make the sensor take the firmware from a local folder. [image] [image]
>
> After that, you'll need to restart Z2M since the configuration file has been [...]

Thanks Your solution working :)

beeline09, 2024-01-25:
> [Koenkk/zigbee-OTA#386](https://github.com/Koenkk/zigbee-OTA/pull/386)
> 
> PR to track

Надо бы это добавить в инструкцию по откату назад. Долго ходил вокруг, да около, пока не наткнулся на это.

Psytoshgen, 2024-01-26:
> > [Koenkk/zigbee-OTA#386](https://github.com/Koenkk/zigbee-OTA/pull/386)
> > PR to track
> 
> Надо бы это добавить в инструкцию по откату назад. Долго ходил вокруг, да около, пока не наткнулся на это.

тут все просто.... останавливаем z2m залазим в файл state.json и в latest version пишем (например файл на который хотим называется 1141-0208-01143001-ZMHOC401N) в последней версии прописываем 01143001 и после запуска z2m видим приглашение обновиться.... а вот как обновиться с альтернативной, еще не придумал...

beeline09, 2024-01-26:
> тут все просто.... останавливаем z2m залазим в файл state.json и в latest version пишем (например файл на который хотим называется 1141-0208-01143001-ZMHOC401N) в последней версии прописываем [...]

Да, спасибо, я именно так и сделал) с альтернативной имеется ввиду не вашей?

closed 2024-02-13 by devbis