devbis/z03mmc #35: Zigbee2MQTT Reporting time

nobet78, 2023-10-31:
Why is the set time in the Zigbee2MQTT Reporting panel not taken into account. I set it to 180 and the actual reporting time is random up to 7 minutes. Has anyone encountered this?

fryc88, 2023-11-01:
Mine doesn't seem to be accepting reporting time changes, temp, humidity and battery are reporting after bang on every 5 minutes, which is draining battery quite a lot i guess - got 78% left from brand new battery, all flashed at saturday so it's 5th day of using it today. I only flashed one of my whole lot of these termometers to see battery drain, but that doesn't look well. With stock BT firmware they lasted 9-12 months each, with custom ATC they were about 4-5 months...
I've got 1.0.4 firmware now, trying to do OTA to 1.0.5 but they don't mention any battery/reporting changes

wimpie007, 2023-11-01:
my 2 devices report every minute spot on.
killing battery.😥

pvvx, 2023-11-02:
The current version 1.0.5 after "bind clasters" passes all values every 10 seconds.
Changing the parameters of the "report" (times) does not work.

wimpie007, 2023-11-02:
mmm, how can we lower the report frequency?
only with an other firmware?

pvvx, 2023-11-02:
Do not “link clusters” 0x402, 0x405,.., but request these attributes using any timer.

wimpie007, 2023-11-02:
zigbee thermometers only transmit when values are changed, and never very fast like 10 seconds.
I dont need to have such a detailed reports. once every x minutes is fine with me.
the reason to use zigbee? range is superieur because i have zigbee devices everywhere. (BLE requires extra devices...)

pvvx, 2023-11-02:
To change the display, it is also required to wake up all the chips in the thermometer, which does not allow for low consumption in the Zigbee version.

devbis, 2023-11-09:
Fixed applying of received reporting values in https://github.com/devbis/z03mmc/releases/tag/1.0.6.

closed 2023-11-09 by devbis