devbis/z03mmc #37: Add option to display clock

mateuszpawlak, 2023-11-01:
Hi,

Custom Bluetooth firmware ([https://github.com/pvvx/ATC_MiThermometer#configuration](https://github.com/pvvx/ATC_MiThermometer#configuration)) give possibility to display also clock (in loop). Is it possible to add this feature to Zigbee firmware?

andershagenhansen, 2023-11-02:
Is it possible to also remove the smiley/comfort indicator ?

Thanks in advance!

devbis, 2023-11-02:
@andershagenhansen you can compile firmware with this line commented.
https://github.com/devbis/z03mmc/blob/master/src/app_cfg.h#L189

There is currently no way to disable it in runtime

andershagenhansen, 2023-11-02:
> @andershagenhansen you can compile firmware with this line commented. https://github.com/devbis/z03mmc/blob/master/src/app_cfg.h#L189
> 
> There is currently no way to disable it in runtime

I will try that - but how do I flash it to my already flashed Zigbee device? :-)

I don't have USB-TTL or resistor available :-)

devbis, 2023-11-02:
You can change fw version and use local OTA for zigbee2mqtt. But in some cases it may brick you sensor (until flashing again via UART). So getting USB-UART is desirable for customizing it.

andershagenhansen, 2023-11-03:
@devbis  it worked with a newly-compiled firmware, and changed the firmware version numbers. Thanks a lot, for your support!

cook1emr, 2023-11-19:
> @devbis it worked with a newly-compiled firmware, and changed the firmware version numbers. Thanks a lot, for your support!

Hi, Is your firmware variation available anywhere? I would also like these options