devbis/z03mmc #39: Rollback guide to bluetooth

raknam, 2023-11-01:
Is it possible to get a guide to rollback to the bluetooth firmware? Maybe via Zigbee OTA?

IvanGavrilenko, 2023-11-03:
You can get original or modified Bluetooth firmware [here](https://github.com/pvvx/ATC_MiThermometer#firmware-binaries) and flash it via USB to UART

devbis, 2023-11-09:
Added a paragraph about reverting to BLE over UART

closed 2023-11-09 by devbis

pvvx, 2023-11-11:
[Zigbee OTA firmware files for recovery in BLE](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota)

devbis, 2023-11-11:
@pvvx Are these firmwares tested? Can I recomment them in the project readme?

pvvx, 2023-11-11:
This is a typical firmware with headers for Zigbee OTA.

devbis, 2023-11-11:
I made such experiments it they failed in 50%. What would happen if BLE firmware is written to address 0x40000?

pvvx, 2023-11-11:
Fixed and tested.