devbis/z03mmc #41: How to flash firmware with USB to UART for MJWSD05MMC?

weichenw, 2023-11-02:
Hi,

I am trying to flash via USB for MJWSD05MMC but have been unsuccessful. Based on the PCB connection diagram, I connected the adapter Txd with 1k ohm to the rxd of MJWSD05MMC. Still no luck. Any pointers?

Thanks
Wei

devbis, 2023-11-02:
This firmware doesn't support MJWSD05MMC yet. You need to add screen management and (possibly) define different i2c group for your device.
But I suppose the connection should be similar to 03mmc:

1. GND
2. 3v3
3. TX --> 1kOhm -> SWS (near rx on the board)
4. RX --> SWS directly

[image]

weichenw, 2023-11-02:
@devbis Thanks.

I tried to connect it as you mentioned and flash the [original firmware](https://pvvx.github.io/MJWSD05MMC/fullflash_MJWSD05MMC.zip), but no luck, put the battery in doesn't boot up. I think it's bricked.

[image]

Thanks again for your help.

devbis, 2023-11-02:
I haven't tried your model but I believe it is impossible to brick tlsr82xx with software manipulation.
Use 'ea' command to erase flash and try again with BLE firmware

weichenw, 2023-11-03:
Thanks, I will give that ago.

MaticSulc, 2025-01-15:
I don't have a separate flasher, but I'm using my Arduino board to read/write UART. I've connected to the pins according to the sketch above (GND, 3.3V, TX to SWS via 1kOHM, RX to SWS).

I'm using https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html and getting this:
```
9:43:28 AM: USB-COM opened.
9:43:52 AM: File was selected, size: 524288 bytes
9:44:19 AM: Reset DTR/RTS (100 ms)
9:44:19 AM: Activate (5 sec)...
9:45:20 AM: Write 524288 bytes in to Flash...
```

When "Activate" appears, I remove the 3.3V supply that I'm feeding from Arduino for 1-2 seconds and plug it back in - but the "Write 524288 bytes in to Flash..." usually appears only a minute or two later (and never ends, I've let it run for 1-2 hours).

Am I missing something? I hope I don't have to go and buy a specific flasher just for this.

donihalim, 2025-07-12:
Hi, I just succesfully flash MJWSD05MMC(EN) using the USBCOM flasher with usb-to-ttl.

Wiring:
UTL GND  -->  -VBAT
UTL 3.3V  -->  +VBAT
UTL TXD  -->  SWS

Flashing:
1. Select USB-COM
2. Baud rate 460800, Activation time 5sec
3. Select the original firmware
4. Erase all flash (wait till complete)
5. Write to Flash (wait till complete)

That is how I flash the device after manage to brick(won't turn on) it. Hope this help.