devbis/z03mmc #42: Is possible to add support for IKEA-VINDRIKTNING?

OXOTH1K, 2023-11-03:
Тут недавно был опубликован [проект](https://github.com/smartboxchannel/IKEA-VINDRIKTNING_ZIGBEE). Получится ли этот датчик скрестить с Xiaomi термометром? В теории выглядит реально.

devbis, 2023-11-03:
В теории - да, если найдётся энтузиаст, который модифицирует прошивку. У меня нет икеевского датчика и на данный момент заниматься этим неинтересно.

Habbie, 2023-11-03:
the VINDRIKTNING does not have any wireless hardware at all

devbis, 2023-11-03:
@Habbie the idea is to solder the board to uart from that sensor, read packets with pm2.5 measurements, and export them to zigbee network in appropriate clusters.

Habbie, 2023-11-03:
oh! that could totally work! Let me know if you need help or want me to test, I wrote https://esphome.io/components/sensor/pm1006.html including the code, and I have spares of both the VINDRIKTNING and the 03MMC :)

Habbie, 2023-11-03:
this would also solve the battery life issue ;)

Ctrl-F4, 2023-11-04:
[ PTVO firmware](https://ptvo.info/zigbee-configurable-firmware-features/external-sensors/particle-concentration-sensors-pm1006k-pmsx003-pms5000/) seems to support this sensor.

devbis, 2023-12-25:
I wrote PoC firmware to add IKEA Vindriktning to zigbee network using lywsd03mmc board.
Here is the compiled firmware https://github.com/devbis/z03mmc/actions/runs/7323760450

And here is a short (text) description for soldering https://github.com/devbis/z03mmc/tree/vindriktning#ikea-vindriktning-upgrade

Shortly, you need DC-DC from 5V to 3V and some wires. 

@Habbie @OXOTH1K

OXOTH1K, 2023-12-26:
Great work! Thanks, @devbis

guzzard, 2024-11-12:
Should Vindriktning REST really be connected directly to 3.3v? It's a data line that should go to some data in on the 03mmc right?
At least on the board I have B1 is directly connected to 3.3v. Or am I misunderstanding the written instructions?

_From https://github.com/devbis/z03mmc/tree/vindriktning?tab=readme-ov-file#ikea-vindriktning-upgrade
`VDRKN REST --> 03MMC B1 pin`_

_In case of revisions B1.5/B1.6 you need to detach 2 pins (or just the second right bottom) from the board. There should be no contact with B1 and any pin on the UART display chip._

[image]

B1 (directly connected to 3.3V
[image]

devbis, 2024-11-13:
No, it should be a gpio pin. Maybe it is p11 here?

devbis, 2024-11-14:
https://github.com/devbis/z03mmc/blob/vindriktning/src/board_8258_03mmc.h#L44

Yes, PB7/RX is used  to receive data from the sensor. You should use P11. I also used rev 1.5 for converting, but can't find the device.

And yes, you need to lift up 1-2 pins on the bottom right side of the display chip (to free P11)

guzzard, 2024-11-23:
> https://github.com/devbis/z03mmc/blob/vindriktning/src/board_8258_03mmc.h#L44
>
> Yes, PB7/RX is used to receive data from the sensor. You should use P11. I also used rev 1.5 for converting, but can't [...]

Working great, thanks!

4thgenlude, 2025-11-15:
I already flashed my lywsd03mmc to use zigbee. Is there any way to update the firmware on it OTA to get the Vindriktning’s PM 2.5 readings? Or is a USB - UART manual flashing the only option?

guzzard, 2025-11-16:
> I already flashed my lywsd03mmc to use zigbee. Is there any way to update the firmware on it OTA to get the Vindriktning’s PM 2.5 readings? Or is a USB - UART manual flashing the only option?

Yeah it's possible to OTA update using for example Zigbee2MQTT with a [Local OTA index and firmware files](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files). I'm sure there are other methods too using ZHA or others, but I only used Zigbee2MQTT..

devbis, 2025-11-17:
@4thgenlude yes, it is possible.

But you need to mangle the header of the zigbee OTA file. The bin should be vindriktning, but the image id  and manufacturer is that are used in your z03mmc 

The python script is here https://github.com/devbis/z03mmc/blob/master/tools%2Fmake_ota.py

4thgenlude, 2025-11-17:
> [@4thgenlude](https://github.com/4thgenlude) yes, it is possible.
>
> But you need to mangle the header of the zigbee OTA file. The bin should be vindriktning, but the image id and manufacturer is that [...]

Thanks! I got it to work OTA via Z2M. Now i just need to wire it all up