devbis/z03mmc #43: Not receiving data in ZHA

gnicolasb, 2023-11-03:
Hello,

I flashed my LYWSD03MMC using an UART. After that, I reset the device and I could configure it using the ZHA integration. ZHA read received data first time, but after that there have been no more updates. Are there any configuration that I should consider?

GooG2e, 2023-11-05:
Have the same problem
But I receive temperature without any problem and humidity not updating at all

devbis, 2023-11-09:
Default reporting values were adjusted in version https://github.com/devbis/z03mmc/releases/tag/1.0.6

Could you please try it with ZHA?

tunip, 2023-11-12:
Flashed 2 LYWSD03MMC with 1.0.6. Behavior not changed. Still very unreliable with ZHA.

GooG2e, 2023-11-14:
I updated to 1.0.6 and receiving data become stable