devbis/z03mmc #47: update interval

YanchikAmb, 2023-11-06:
Hi! How I can change update interval in ZHA?

devbis, 2023-11-06:
1. The default behavior is fixed in https://github.com/devbis/z03mmc/commit/87b9708b01b726751d7b3b4eed14d53102d31798 and will be in the next release. It will notify in 10 seconds if the value changed and in 5 minutes if value change didn't change more than the limit. For temp it is 0.1º, for humidity - 0.5%. 

2. I don't know how to set up reporting in ZHA, probably https://github.com/mdeweerd/zha-toolkit could help.

pvvx, 2023-11-16:
> The default reporting behavior is fixed in https://github.com/devbis/z03mmc/commit/ec3f46c750dd31bb2b62b25b80c866a6ad4a0255 and will be in the next release. It will notify in 10 seconds if the value changed and in 5 minutes if value change didn't change more than the limit. For temp it is 0.1º, for humidity - 0.5%.

Does not work. Only one thing is transmitted - temperature, humidity or battery.
Reason: The same [timer is used for all reports](https://github.com/devbis/z03mmc/blob/master/src/reporting.c#L80).