devbis/z03mmc #48: Support for other Telink based devices?

giggo, 2023-11-07:
I flashed the firmware to a [Xiaomi Mijia BLE MHO-C401](https://pvvx.github.io/MHO_C401/) and Zigbee works fine.  The display obviously doesn't. Do you have plans to support the other devices that are supported in @pvvx 's firmware (https://github.com/pvvx/ATC_MiThermometer)?

devbis, 2023-11-07:
Yes, I have such plans, but it's not a top priority since I don't have such devices. :) 
Does temperature and humidity are detected correctly?

pvvx, 2023-11-07:
> Are temperature and humidity detected correctly?

Defective or fake sensors are installed on variants С401 from Azarton. The data is sent to a different I2c address (0x00).

pvvx, 2023-11-11:
Added CGDK2 in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

pvvx, 2023-11-13:
Added test version MHO-C401(N), where N is release from 2022.
MHO-C401 have 3 global hardware modifications...
[image]