devbis/z03mmc #50: An alternative version flash the Zigbee firmware in Xiaomi LYWSD03MMC in TelinkMiFlasher from version 6.7.

pvvx, 2023-11-08:
**An alternative version flash the Zigbee firmware in Xiaomi LYWSD03MMC in TelinkMiFlasher from version 6.7.**

To flash the Zigbee firmware, use a Google Chrome, Microsoft Edge or Opera Browser.
	1. Go to the [Over-the-air Webupdater Page TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
	2. If using Android, Windows, Linux: Ensure you enabled "experimental web platform features". Therefore copy the according link (i.e. chrome://flags/#enable-experimental-web-platform-features for Chrome), open a new browser tab, paste the copied URL. Now sten the _Experimental Web Platform features_ flag to _Enabled_. Then restart the browser.
	3. In the Telink Flasher Page: Press `Connect:`. The browser should open a popup with visible Bluetooth devices. Choose the according target device (i.e. LYWSD03MMC) to pair.
	4. After connection is established a _Do Acivation_ button appears. Press this button to start the decryption key process.
	5. Now you can press the Zigbee Firmware button to directly flash the `Zigbee Firmware`:<br>[image]<br>Alternatively you can choose a specific firmware binary (i.e. the original firmware) via the file chooser
	7. Press Start Flashing. Wait for the firmware to finish.
	8. The device should now show up in your Zigbee bridge (If joining is enabled, of course). If it doesn't, reinsert the battery and/or short the RESET and GND contacts on the board for 3 seconds.

A brief description of the installed version of the Zigbee firmware is in this [fork](https://github.com/pvvx/z03mmc).
In the future, you can update the [Zigbee firmware to the version from devbis](https://github.com/devbis/z03mmc).

adamlc, 2023-11-08:
I did try this with a fresh device out of the box. The firmware did flash OK with no errors.

I waited awhile and it didn't show up in z2m. So I pulled the battery and then the device appears to be dead, could well be related to #29. This new device is B2.0 which I believe has some outstanding issues.

pvvx, 2023-11-11:
[Zigbee OTA firmware files for recovery in BLE](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota)

heggi, 2023-11-24:
Прошил девайс LYWSD03MMC флешером по ссылке выше (версия флешера 7.0)
Сделал все по инструкции. Все ок.
Закинул в папку для OTA последнюю прошивку 1141-0203-10063001-z03mmc.zigbee
Запускаю обновление, но ничего не происходит. В логах файл прошивки определяется как 

`ImageKey(manufacturer_id=4417, image_type=515): /config/zigpy_ota/1141-0203-10063001-z03mmc.zigbee, version: 268840961, hw_ver: (None, None), OTA string: `

Сам девайс говорит о себе следующее: 
`OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=4417, image_type=522, current_file_version=17969153, hardware_version=None, model='LYWSD03MMC'`

Учитывая, что image_type не совпадает, возник вопрос. А что именно мне залил TelinkMiFlasher ?

closed 2023-11-25 by pvvx