devbis/z03mmc #55: OTA update issues with Zigbee2MQTT

booroondook-eng, 2023-11-09:
Имеется 7 датчиков (5 с ревизией 1.4 и 2 с 1.7).
Никак не могу их обновить по ОТА, используя соответствующую возможность Z2M.
Пробовал и локальное обновление (файл .zigbee в папке с данными), и через Интернет.
Лишь один раз (по-моему, это было обновление с 1004 на 1005) два датчика (один 1.4 и один 1.7) успешно обновились. При этом процедура обновления заняла тогда около 25 минут у каждого датчика.
Попытки обновить другие датчики неудачны: сначала запрос обновления. Обновление есть. Затем запуск обновления. Где-то только через минуту появляется серый прогресс-бар. Начинается обновление. Прогнозируемое время - может быть 20 минут, может быть 50 минут, а может быть и несколько часов. Обновление идет ужасно медленно и дойдя до некоторого значения (бывало 2% , бывало 17%) обрывается. Датчик при этом остается рабочим и продолжает сообщать старую версию прошивки и наличие обновления. В некоторых случаях обновление просто останавливается, прогресс не нарастает. Прогнозируемое время обновления меняется на "N/A". Перезагрузка Z2M (в том числе, и рестартом сервиса) не помогает - после перезагрузки рядом с проблемным датчиком все также высвечивается пара процентов обновления и "Оставшееся время N/A".
P.S. Обновление датчиков делаю по очереди. Одновременно пытаюсь обновить только один датчик.

kaizer666, 2023-11-10:
Такая же проблема. Останавливается обновление с ошибкой "Update of 'Датчик температуры и влажности 2' failed (Timeout: device did not request any image blocks)", при этом никакого прогресса нигде не вижу. Подключен через zigbee2mqtt

Ccrrocus, 2023-11-10:
Такая же проблема. В Настройках 3 датчика (всего их 3) уже 6 часов в "Установке". При этом в z2m в списке нет значка обновления, как происходило ранее до этой версии. Перегрузка z2m да и HA ни к чему не привело.

kaizer666, 2023-11-10:
> Такая же проблема. В Настройках 3 датчика (всего их 3) уже 6 часов в "Установке". При этом в z2m в списке нет значка обновления, как происходило ранее до этой версии. Перегрузка z2m да и HA ни к чему не привело.

Пробовали добавлять новый конвертер? Мне помогло именно это действие. НО я дождался когда процесс прошивки прервётся (в моём случае с ошибкой). После этого обновил конвертер и перезагрузил z2m. И обновление встало

Ccrrocus, 2023-11-10:
> НО я дождался когда процесс прошивки прервётся (в моём случае с ошибкой).

Пока нет ошибки и прерывания установки. Датчики работают, данные от них не прерываются. Подожду.

kaizer666, 2023-11-10:
@Ccrrocus пробовали залить файл [прошивки](https://github.com/devbis/z03mmc/releases/download/1.0.6/1141-0203-10063001-z03mmc.zigbee)  на сервер и обновиться с него? Я так розетку от Aqara обновлял в своё время, потому что по воздуху не хотела обновляться.

Ccrrocus, 2023-11-10:
> пробовали залить файл [прошивки](https://github.com/devbis/z03mmc/releases/download/1.0.6/1141-0203-10063001-z03mmc.zigbee) на сервер и обновиться с него?

Обновить "во что бы то ни стало" задачи не стоит. Считаю, что обратная связь с разработчиком "дороже" единственного обновления.

Ccrrocus, 2023-11-10:
[image]
Судя по этому, датчик и не думает обновляться)

devbis, 2023-11-10:
Датчик сам не принимает решение обновиться, только по команде от координатора

Ccrrocus, 2023-11-10:
[image]
Судя по картинке, он решение обновиться принял.
Остановил z2m, заменил конвертер на новый, запустил z2m. Изменений пока не вижу.

kaizer666, 2023-11-10:
@Ccrrocus я вообще отключил этот сенсор. Он по сути показывает наличие обновления. а не сам процесс обновления. Обновлять только через интерфейс z2m
[image]

Ccrrocus, 2023-11-10:
Спасибо. Попробую через OTA z2m.Хотя прошлые обновления прилетали без проблем.

korttoma, 2023-11-10:
I have 3 of these that were running firmware version 1.0.5 conected to z2m. One of them detected that the new 1.0.6 vas available so I triggered OTA for it. The update failed at bout 80% with "aborted by device". Several hours later the next senor had noticed that 1.0.6 is available so then I triggered update and for this one it wasa success.
Then I again triggered update for the first sensor and now allso this one was successfully updated. The 3rd and last sensor has not yet noticed that 1.0.6 is available. Even if I use the "Search" button. Guess I just have to wait.

propi62, 2023-11-10:
ota worked perfect
[image]

booroondook-eng, 2023-11-11:
В-общем, научился я обновлять датчики через ОТА в Zigbee2MQTT. Читайте, кому надо, мою "инструкцию".
Основное правило - одновременно обновляем только один датчик. Пока не обновили предыдущий, за следующий не берёмся.
Правило второе - если при попытке обновления через сколько секунд вылезает красная плашка с информацией о том. что датчик "чего-то там не то", то идем к датчику, передергиваем ему батарейку. А чтобы время не терять, пока он прочухается, можем обновить другой датчик.

Для обновления жмем кнопку "Обновить прошивку устройства". Ждем (иногда долго) появления прогресс-бара со счетчиком процентов и информацией об оставшемся времени. Оставшееся время - величина чисто условная - вам может быть показано 20 секунд, а по факту прошивка растянется на полчаса и более.
Терпеливо ждём окончания процесса. Тут либо всё дойдет до 100% (и тогда можно праздновать), либо оборвется на каком-то количестве процентов и остановится. Возможно, при этом вы увидите пустой прогресс бар и оставшееся время "N/A". Или же бар, заполненный совсем чуть-чуть. Всё равно ждём до того момента, пока снова не возникнет кнопка "Обновить прошивку устройства". Даже не пытайтесь как-то повлиять на процесс - просто терпеливо ждите, даже если придется ждать целый час или более. Можете в это время заняться другими делами.
В-общем, когда кнопка появилась, снова жмём на нее. Снова через время появится прогресс-бар, но уже не "нулевой", а с процентами (например, 25%, 57%, 80% и т.д.). Терпеливо ждем нарастания процентов. В какой-то момент снова может произойти "срыв". Если так, то опять жмем кнопку "Обновить..."
В-общем, таким способом, не спеша, как бы "поэтапно" прошиваем по очереди все наши датчики. Времени на это уйдет уйма может быть, даже несколько дней. Но никто ведь не заставляет вас все это время пялиться в экран. Сходите погуляйте, займитесь домашними делами и т.д.

pvvx, 2023-11-11:
Zigbee OTA прошивка размером к 128 килобайт происходит за 677 секунд.
При этом потребление термометром около 1 мА. Т.е. требуется энергия в 5.4 мАч
Время старта и конца Zigbee OTA прошивки из лога шлюза Zigbee2BLE:
```
23-11-11 01:52:39.034216 recv<--:55 80 00 00 04 15 01 90 00 00 aa 
...
23-11-11 02:03:56.293907 recv<--:55 80 43 00 0b 35 b2 0a a4 c1 38 88 83 3e 5d f0 80 aa 
```