devbis/z03mmc #56: SyntaxError during build in file make_ota.py

Juul, 2023-11-11:
This happens both in latest from git following the build instructions from the readme and using the 1.0.6 release. System is Ubuntu 20.04 with Python 3.8.10 and cmake 3.16.3

The error:
```
Scanning dependencies of target z03mmc.bin
Firmware patched!
[100%] Built target z03mmc.bin
Scanning dependencies of target z03mmc.zigbee
  File "z03mmc-1.0.6/tools/make_ota.py", line 14
    with (open(args.input_file, 'rb') as bin_file):
                                      ^
SyntaxError: invalid syntax
make[3]: *** [src/CMakeFiles/z03mmc.zigbee.dir/build.make:57: src/CMakeFiles/z03mmc.zigbee] Error 1
make[2]: *** [CMakeFiles/Makefile2:96: src/CMakeFiles/z03mmc.zigbee.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:103: src/CMakeFiles/z03mmc.zigbee.dir/rule] Error 2
make: *** [Makefile:118: z03mmc.zigbee] Error 2
```

Juul, 2023-11-11:
Looks like it works with Python 3.9

Juul, 2023-11-11:
Created pull request #57 to add a note about this to the readme

devbis, 2023-11-11:
Syntax is fixed in https://github.com/devbis/z03mmc/commit/49a8e4f796d2b8ab3e9f360b7885a4c1256ace6a

closed 2023-11-11 by devbis