devbis/z03mmc #58: Support for Conbee II?

Avils, 2023-11-11:
Is there any chance that I can pair the xiaomi sensor through conbee gateway?

Thanks.

devbis, 2023-11-11:
Does conbee II work with zigbee2mqtt or ZHA? 
This firmware uses standard clusters to work with multiple coordinator chips.

c4software, 2023-11-12:
I confirm, it's already working with Conbee II and the latest zigbee2mqtt release.

Avils, 2023-11-12:
> I confirm, it's already working with Conbee II and the latest zigbee2mqtt release.

How did you pair it? My Conbee gateway can't discover the sensor.

c4software, 2023-11-12:
Like my other zigbee devices. Open pairing in zigbee2mqtt and the device is joining the network like usual.

Avils, 2023-11-12:
> Like my other zigbee devices. Open pairing in zigbee2mqtt and the device is joining the network like usual.

Oh, I was trying to pair it in the Phoscon app.