devbis/z03mmc #60: update of failed (update failed with reason: 'invalid image')

PetrMa, 2023-11-11:
Hello,

today I made a convesrion of my lywsd03mmc from BT to Zigbee.
Everything works fine, device correctly reported all statuses but only one issue is with OTA.
The Z2M indicates tat new update is available but all attempts to update end with an error "update of failed (update failed with reason: 'invalid image')"

Currently the device firmware is:
Firmware date: 20231108
Firmware version: 3001-0101

The HA see this information:
Installed version: 16855041
Last version: 268840961

Any ideas how to fix this issue please?

Thanks

devbis, 2023-11-11:
Probably you've installed the image from pvvx. They are incompatible, according to the values you have.

PetrMa, 2023-11-11:
> Probably you've installed the image from pvvx. They are incompatible, according to the values you have.

Yes, probably it is a source of the issue. Is there any way how to fix it by OTA as I don't have UART.

devbis, 2023-11-11:
You may try this file as OTA fw to update to 1.0.6.
[1141-020a-10063001-z03mmc.zip](https://github.com/devbis/z03mmc/files/13326510/1141-020a-10063001-z03mmc.zip)

PetrMa, 2023-11-11:
How can I update the thermometer by this file with OTA? The device is already on Zigbee firmware so I'm not able to use ovvx again.
Is there any way how to push it by Z2M manually?

devbis, 2023-11-11:
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

You need to add this file to z2m data folder and set up local OTA index.

PetrMa, 2023-11-12:
Thanks, I made a local json update file with:

[
    {
        "url": "1141-020a-10063001-z03mmc.zigbee",
        "force": true
    }
]

but still ends with the same error message:
Error 2023-11-12 10:36:59Update of '0xa4c1383a66a98ccc' failed (Update failed with reason: 'invalid image')

Deekirs, 2023-11-12:
I have 3 sensors. 2 are updated perfectly, one with problems. Crashes with the same error. At the same time, it was previously updated via OTA.

devbis, 2023-11-13:
> It is possible to make some OTA image which can replace the pvvx version please?

@PetrMa It is the firmware with patched header to update from pvvx's image. Did you restart z2m? 
Did you add local ota record to the configuration file?

PetrMa, 2023-11-13:
@devbis Yes and Yes. The firmware in in my zigbee2mqtt folder, configuration file includes the 
ota:
  ikea_ota_use_test_url: false
  disable_automatic_update_check: false
  zigbee_ota_override_index_location: my_ota.json

and my_ota.json includes:
[
    {
        "url": "1141-020a-10063001-z03mmc.zigbee",
        "force": true
    }
]

as you can see in attached screenshot. Z2M was restarted but the update finished with the same error mentioned above.
[image]

o, 2023-11-14:
Hi! first of all thanks for the hard work. 
I recently flashed a LYWSD03MMC works pretty well. Yesterday I saw an OTA update in z2m interface and tried to update but I am getting the same error mentioned in the previous posts.

for the context: using z2m with sonoff P dongle, firmware version 3001-1005, device LYWSD03MMC

```
 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Studio Refrigerator temperature sensor"},"error":"Update of 'Studio Refrigerator temperature sensor' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"rsj0s-1"}'
```

PetrMa, 2023-11-14:
I´am also on Z2M with Sonoff P dongle, but on another firmware 3001-0101

PetrMa, 2023-11-20:
> > It is possible to make some OTA image which can replace the pvvx version please?
> 
> @PetrMa It is the firmware with patched header to update from pvvx's image. Did you restart z2m? Did you add local ota record to the configuration file?

Hi @devbis 
is there any way how to fix this issue please? 

Thank you so much, Petr

SPEC1AL1ST, 2023-11-20:
Go to /homeassistant/zigbee2mqtt/configuration.yaml
Add:
ota:
  ikea_ota_use_test_url: true
  zigbee_ota_override_index_location: my_index.json

Add file my_index.json to  /homeassistant/zigbee2mqtt/
Add to file:
[
    {
        "url": "1141-020a-10063001-z03mmc.zigbee",
        "force": true
    }
]
  
Upload file from zip archive 1141-020a-10063001-z03mmc.zigbee to /homeassistant/zigbee2mqtt/
[1141-020a-10063001-z03mmc.zip](https://github.com/devbis/z03mmc/files/13413109/1141-020a-10063001-z03mmc.zip)

Restart z2m addon

Go to OTA and click check update - Update it :)

devbis, 2023-11-20:
@PetrMa  You config looks correct, but I don't use HA in docker, so I'm not sure where the config files for z2m are located. I suggested @SPEC1AL1ST to flash this file and it worked for his setup. Maybe the config folder is /homeassistant instead of /config? It looks like z2m just don't see changes.

PetrMa, 2023-11-20:
@SPEC1AL1ST  as you can see in my previous posts, I made all these steps and the result is:
Error 2023-11-12 10:36:59Update of '0xa4c1383a66a98ccc' failed (Update failed with reason: 'invalid image')

and also other user in this thread has the same issue. The question is what was you previous firmware version. I´am on 3001-0101 now

SPEC1AL1ST, 2023-11-20:
> @PetrMa You config looks correct, but I don't use HA in docker, so I'm not sure where the config files for z2m are located. I suggested @SPEC1AL1ST to flash this file and it worked for his setup. Maybe the config folder is /homeassistant instead of /config? It looks like z2m just doesn't see changes.

Im use File editor addon
[image]

SPEC1AL1ST, 2023-11-20:
> and also other user in this thread has the same issue. The question is what was you previous firmware version. I´am on 3001-0101 now

Previous:
[image]
Now:
[image]

PetrMa, 2023-11-20:
@SPEC1AL1ST @devbis 
I tried to rename the zibbee file in my_ota.json to "zigbe" 
[
    {
        "url": "1141-020a-10063001-z03mmc.zigbe",
        "force": true
    }
]

and the result is:
Update of '0xa4c1383a66a98ccc' failed (ENOENT: no such file or directory, open '/config/zigbee2mqtt/1141-020a-10063001-z03mmc.zigbe')

With the correct update file name I´ve go the error:
Update of '0xa4c1383a66a98ccc' failed (Update failed with reason: 'invalid image')

So I can confirm that the configuration for manual update works fine and Z2M really use the correct file for update but there is some issue to make an update from version 3001-0101. @SPEC1AL1ST had different version.

SPEC1AL1ST, 2023-11-20:
.zigbe

not .zigbe

.zigbee

PetrMa, 2023-11-20:
> .zigbe
> 
> not .zigbe
> 
> .zigbee

Yes, I know ..... I renamed the file in the json file only to test that Z2M takes the correct file from the correct location. Please, don´t ignore my previous posts. You can check my settings any you will see that I use the correct OTA settings.

devbis, 2023-11-20:
@PetrMa It looks like your device num is neither 0a nor 03. Or other internal checks do not pass, it's difficult to figure out.
Please try local update to pvvx's firmware https://github.com/pvvx/ZigbeeTLc/tree/master/bin while you have his binary. Maybe it will work. If no, the only way I can advise is to switch is flashing over UART.

PetrMa, 2023-11-20:
[image]

and the content of my_ota.json is:
[image]

and the reult:
[image]

PetrMa, 2023-11-20:
> @PetrMaZdá se, že číslo vašeho zařízení není ani 0a, ani 03. Nebo jiné interní kontroly neprošly, je těžké to zjistit. Zkuste prosím místní aktualizaci firmwaru pvvx https://github.com/pvvx/ZigbeeTLc/tree/master/bin , dokud máte jeho binární soubor. Možná to půjde. Pokud ne, jediný způsob, jak mohu poradit, je přepnout blikání přes UART.

Do you know which zigbee file I need to use please?
[image]

PetrMa, 2023-11-20:
> @PetrMa It looks like your device num is neither 0a nor 03. Or other internal checks do not pass, it's difficult to figure out. Please try local update to pvvx's firmware [...]

Do you know which zigbee file i need to use for update of ppvx firmware?
[image]

devbis, 2023-11-20:
z03mmc

PetrMa, 2023-11-20:
The same error. I´am not sure if his zigbee file is usable woth Z2M because pvvx uses only ZHA. So maybe his zigbee file is not fully compatible with Z2M

devbis, 2023-11-20:
Unfortunately, he deleted the repo where your firmware is and it's hard to distinguish the image type.

PetrMa, 2023-11-21:
@devbis FYI: this is the problem with old pvvx firmware. So the USB UART is the only way at this time with this specific firmware.
https://github.com/pvvx/ZigbeeTLc/issues/11#issuecomment-1819887341

closed 2023-11-21 by PetrMa

devbis, 2023-11-21:
@PetrMa Thanks for updates!