devbis/z03mmc #61: Возможность изменять информацию на экране

LaSScaR, 2023-11-13:
Планируется ли в будущем возможность изменять информацию на экране?
К прмиеру часы, дата, убрать смайлик, показ заряда батареи, возможность изменять частоту передачи информации от датчик к стику.
Заранее спасибо.

devbis, 2023-11-13:
На данный момент часы, дата, показ заряда батареи не в разработке.
Изменять частоту передачи вы можете в настройках отчётов для каждого кластера и атрибута в отдельности
Отключить смайлик можно собрав прошивку с отключённым флагом DISPLAY_SMILEY.
Если для ваших нужд требуется расширенная функциональность экрана, рекомендую использовать Bluetooth прошивку от pvvx.

LaSScaR, 2023-11-13:
[image]
Извиняюсь за дурной вопрос., частоту передачи вот здесь? На скрине обвёл красным, значения в секундах, верно?

mathieu5798, 2024-01-14:
Hello, I'm sorry to up this issue.
I can't find how to deactivate the smiley face... Where can I found the DISPLAY_SMILEY you're talking about ?
I'm using zigbee2mqtt.
Thanks