devbis/z03mmc #64: Question about Zigbee Tuya

klu16, 2023-11-16:
Hi,

my parents have some LYWSD03MMC which I flashed with Zigbee firmware. Instead of a Home Assistant, they have simply a Tuya gateway (all available via Tuya Smart app).

After flashing the Zigbee firmware, they paired with the app just fine and are visible. BUT... I wonder if it is possible to display the temperature more accurately in the app with decimal? I mean, the sensor is showing on the LCD screen e.g. 20,8'C but the app is showing round 20'C. Is this a Tuya app-specific thing or can it be changed somehow to display the temp with decimal?

Thank you for your awesome work! Much appreciated :)

Screenshot:
[image]

devbis, 2023-11-16:
Hi! That's cool it can pair with Tuya gateway as it might have a white list of devices. As for the values, both temperature and humidity are stored and reported with decimals up to 0.01, so the rounding is done by the app. Unfortunately, i don't have Smart life app and can't check options. 

The values are literally reporting as 2034 (20.34°), so the only possible culprit is the app itself

klu16, 2023-11-16:
Thanks Ivan for your valuable feedback! I thought that the culprit might be the app itself, but I wanted to make sure.
PS. I went through all the settings within the app and there is no possibility to change the display values. Maybe I will write to Tuya support and make a suggestion for improvement.
Thanks again! :)

klu16, 2023-11-17:
By the way, this is the answer I received from Tuya:

"Dear customer,
Thank you for contacting us.
We apologize for the inconvenience,The numerical system of the device control panel in the APP is configured by the device manufacturer during the development process. As an APP, we do not have the authority to change it.As an IoT platform, Tuya Smart provides a development platform for product manufacturers or developers,  However, we are unable to add functions to individual products.  Developers are responsible for this configuration.

To clarify your requirements, you should contact the device's developer or manufacturer.

Powered by Tuya (PBT) is a mark of interconnectivity across different brands and categories, which means that this product is developed based on the Tuya Smart IoT platform."

Just FYI :)

devbis, 2023-11-17:
Maybe there is a way to register third-party device in Tuya database and configure display values but I'm not sure it is possible. Let me know, please, if you could find a way to change settings.

klu16, 2023-11-17:
I found this:
https://developer.tuya.com/en/docs/iot/shift-temperature-scale-follow-the-setting-of-app-account-center?id=Ka9qo7so58efq

I created a Cloud project where the temp sensors are visible, but there isn't any way to change the scale:
[image]

The article above mentions to create a product, but I am not sure how to proceed further there... It looks like this:

[image]

I am really NOT a developer, so it's a bit difficult for me to understand some things :(

pl4za, 2025-06-21:
@klu16 I'm struggling to get the sensors to send updates to my tuya gateway. What fw version are you running and gateway do you have? Thanks.