devbis/z03mmc #66: OTA fails with wrong imageType

oleksii-kalinin, 2023-11-16:
Trying to update with z2m and got an error:
```
"error":"Failed to check if update available for 'Climate Living Room' (No image available for imageType '522')"
```

However no image for imageType: 522 is available in the OTA repo. Manually overriding 515 to 522 imageType didn't work as well
```
"error":"Update of 'Climate Living Room' failed (Image type mismatch)
```

pvvx, 2023-11-17:
To restore to BLE, use the Zigbee OTA file with the same number from: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

After running Zigbee OTA, be sure to delete this file or files from the update directories for Zigbee OTA and completely restart z2m/zha. Otherwise, with the new Zigbee firmware, this update in BLE will work again.

To update Zigbee, use the following files:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin

While the firmware options are in beta testing, I will not post them in the [general lists of automatic updates](https://github.com/Koenkk/zigbee-OTA) for all Zigbee devices.

mondfalter, 2023-11-18:
First of all, thank you for your work!

I have the same issue with one of the two sensors I updated. One was updated directly to 3001-1006 after flashing. The other has firmware version 3002-0103 but can't be updated via OTA.
The same error from above ( no image available for Imagetype '522')

Did I do something wrong when I flashed this sensor?

pvvx, 2023-11-19:
Thermometers with "Manufacturer Code": 4417 or 0x1141, "imageType":
• **0x020a** is a Xiaomi LYWSD03MMC with firmware from [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc),
• **0x0203** is Xiaomi LYWSD03MMC with firmware from [z03mmc](https://github.com/devbis/z03mmc).

mondfalter, 2023-11-19:
So I'm using 2 different firmwares?

Any clues which are preferable? E.g. regarding battery lifetime and future development?
 I've seen some updates on devbis version but couldn't find anything on yours pvvx.

I gathered it's somehow possible to change between them using OTA. I will probably figure that one out.

Guidance for choosing though would be greatly appreciated!

Thanks!

pvvx, 2023-11-19:
> E.g. regarding battery lifetime and future development?

https://github.com/pvvx/ZigbeeTLc/issues/3#issuecomment-1814897450

closed 2023-11-19 by oleksii-kalinin