devbis/z03mmc #68: Bluetooth plant sensors

fryc88, 2023-11-18:
Hi,
Have you got any plans to add/convert plant sensors as well? Xiaomi and couple of other companies make them, they all look and work the same, i think the correct model is HHCCJCYO1HHCC. 
I have quite a few of them, over the BT i had less issues with them than with original LYWSD03MMC but it would be lovely to see them on Zigbee, especially when there's no other zigbee plant sensors available really.

devbis, 2023-11-18:
It would be nice, but according to specs https://www.nordicsemi.com/products/nrf52832 its chop supports only ble and no zigbee. I don't know what is the difference between nordic chips besides their models.

There are several diy models https://www.tindie.com/stores/diyberk/ maybe it could be useful for you.

pvvx, 2023-11-23:
Refer to https://github.com/rbaron/b-parasite