devbis/z03mmc #69: Version 1.06, after re-flashing with B1.6, unable to connect to ZHA.

MeYang901225, 2023-11-18:
"I started by buying three Xiaomi Temperature and Humidity Sensors 2, all flashed with either version 1.03 or 1.04. Later, after an OTA update, I bought a new one. After a failed attempt to flash it, it turned into a brick. I tried using USB to flash it back to ATC 46 and then flashed version 1.06 of z03mmc. However, now it cannot connect to ZHA. I've tried restarting after disconnecting the power, short-circuiting GND, and rebooting, but none of these methods worked to establish a proper connection to ZHA. What should I do?"

devbis, 2023-11-18:
If you have UART, you may erase flash with "ea" command and write the firmware. It should give you a "factory-new" device.

pvvx, 2023-11-19:
Writing any new firmware ([ATC_MiThermometer](https://github.com/pvvx/ATC_MiThermometer) and [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc))  to the “dirty” Flash has nothing to do with the further operation of the thermometer. 

A non-working thermometer can only come out if the firmware writing process was unsuccessful. In such cases, the recording should be repeated.
The “ea” command is faster and if it works, then the subsequent firmware session will not encounter a sleeping chip - the probability of initial agreement on the protocol for transferring the firmware code will increase during subsequent firmware.