devbis/z03mmc #70: Doubt with installation on B1.4 and B1.5 sensor

kikher, 2023-11-18:
Hi, I'm new to this. I have been reading a lot about switching from BLE to Zigbee and I am interested.
I have 2 B1.4 sensors and 4 B1.5 sensors.
I have seen that there are two way to switch versions. One via a transitional firmware (1) and one direct with Flasher version 6.9 (2).

1. https://github.com/devbis/z03mmc
2. https://github.com/pvvx/ZigbeeTLc

Which of the two is more advisable to follow? Can it give me any problem? I have all the sensors in version 1.0.0_0130.

Thank you very much for any help you can offer me.

pvvx, 2023-11-19:
Both firmware versions are in beta testing. Firmware from [devbis](https://github.com/devbis/z03mmc) is in the general update lists for all Zigbee devices and can be updated automatically in some applications.
Firmware from [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc) will not be published in the general update lists until they pass the beta testing stage. Update only manually.

kikher, 2023-11-19:
Thank you very much for your comment, I will use the Firmware from [devbis](https://github.com/devbis/z03mmc).
I'm not very clear about the last part you tell me. I had read that it can only be returned to BLE by USB... but I understand that through my Zigbee platform I can return it to BLE.
I use deCONZ and it has the OTAU Update option. Does this mean that using this function I could bring it back to BLE without a USB USRT?
I have accessed the link [ATC_MiThermometer - zigbee_ota](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) and see several files all ending in .zegbee but not .ota or similar, any of them would work to return the LYWSD03MMC to BLE.
Sorry for asking so much but before I get to flashing it I want to make sure I understand everything right and know if it has a way back if I am wrong about anything.

pvvx, 2023-11-19:
> I have accessed the link [ATC_MiThermometer - zigbee_ota](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) and see several files all ending in .zegbee but not .ota or similar, any of them would work to return the LYWSD03MMC to BLE.

The extension after the file name does not matter. 

> I had read that it can only be returned to BLE by USB... 

I have already made a version of the Zigbee OTA firmware that converts the thermometer back to the BLE version and for the devbis firmware. This file is "1141-0203-99993001-ATC_v46.zigbee".

kikher, 2023-11-19:
Thank you very much I will try it