devbis/z03mmc #71: Sensors stuck

Operrr, 2023-11-20:
I have LYWSD03MMC sensors, I flashed 5 of them with firmware 1.0.6 and after some time all the sensors freeze, stop sending readings and the values ​​on the display do not change. After a power reset they work normally for a while and freeze again. On the stock BT firmware the sensors worked flawlessly

drwalr, 2023-11-20:
also have this problem,
Manual reading in Z2M gives an error in the 'MAC transaction expired' logs

[image]

The connection to the coordinator is good.

oosinnyi, 2023-11-24:
Same problem on latest firmware.

wimpie007, 2023-11-24:
Same issues here. OTA from 1.0.5 to 1.0.6 was successfull, but after a few hours, reporting stopped. (and display freezed)
Pulling battery kicks it on again.

pvvx, 2023-11-25:
This is a particular error when switching the operation of the end device through a router.
For such cases, it is advisable to support a “Poll Management Cluster” on the end device and router.

wimpie007, 2023-11-26:
This is the only device i have (and i have a LOT) that shows this behavior.
I didnt configure a "Poll Management Cluster” for neither of the devices....

pvvx, 2023-12-11:
> So the problem is in both ZHA and zigbee2mqtt or in the zigbee coordinator itself?

I have not observed any of the problems described here on ZHA since the first versions of the firmware.
It is not even possible to reproduce the problems described.

jul1an353, 2023-12-13:
I have the same problem with seven sensors I have just bought, so probably the latest hardware revision. Zigbee2mqtt with SONOFF Zigbee 3.0 USB Dongle Plus coordinator. all sensors freeze after about a day in use. 
I have tried to downgrade to 1.0.5 without success.

Error 2023-12-13 15:24:31 Update of 'wohnen sensor2' failed (Image type mismatch)

my_index.json:
[
    {
        "manufacturerCode": 4417,
        "imageType": 522,
        "url": "1141-0203-10053001-z03mmc.zigbee",
        "force": true
    }
]

I would appreciate any help with downgrading.

pvvx, 2023-12-13:
1. Restore to BLE, use the Zigbee OTA file with https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-0203-99993001-ATC_v46.zigbee
2. Use [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) by pasting the link "https://github.com/devbis/z03mmc/releases/download/1.0.5/1141-0203-10053001-z03mmc.zigbee" into the download option.

pvvx, 2023-12-13:
https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-0203-99993001-ATC_v46.zigbee

Internal Headers:
Zigbee OTA Info: ManCode: 1141, ImageType: 0203, FwVer: 99993001 - APP Release: 9.9 Build: 9.9, Stack Version: 3.0 Release: 0.1

jul1an353, 2023-12-14:
I don't get it.

z2m does not take firmware of Imagetype 515. it only wants 522.

Debug 2023-12-14 11:36:19ZigbeeOTA: getting local firmware file /app/data/1141-0203-99993001-ATC_v46.zigbee
Error 2023-12-14 11:36:19Failed to check if update available for 'flur sensor' (No image available for imageType '522')

z2m my_index.json:
[
    {
        "manufacturerCode": 4417,
        "imageType": 515,
        "url": "1141-0203-99993001-ATC_v46.zigbee",
        "force": true
    }
]

also no luck witch zha

[zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): /config/zigpy_ota/1141-0203-99993001-ATC_v46.zigbee, version: 2576953345, hw_ver: (None, None), OTA string: Telink OTA BLE device
[zigpy.zcl] [0x1427:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=4417, image_type=522, current_file_version=17903617, hardware_version=None, model='LYWSD03MMC'
[zigpy.zcl] [0x1427:1:0x0019] No OTA image is available

i give up with ota and try to downgrade by wire. thanks a lot for the hints anyway. :-)

WickedDurkie, 2024-02-12:
This is still an issue on the latest firmware (1.10).

Fr3di, 2024-02-14:
Have the same issue after updating to v1.10 but only on one device :/