devbis/z03mmc #73: Problems building the firmware (CMAKE)

breaver, 2023-11-22:
When executing this command, 

`cmake -B build -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141`

i receive below error:

```
D:\Downloads\z03mmc>cmake -B build -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
-- Building for: NMake Makefiles
CMake Error at CMakeLists.txt:11 (PROJECT):
  Running

   'nmake' '-?'

  failed with:

   no such file or directory

-- Configuring incomplete, errors occurred!
```

I am running on Windows. Any clue?

pvvx, 2023-11-23:
This assembly is not compatible with [Telink_IoT_Studio](http://wiki.telink-semi.cn/wiki/IDE-and-Tools/Telink_IoT_Studio/).
Install the full MSYS or MinGW package... 
Or use WSL (preferred for this build).

devbis, 2023-11-24:
@breaver I don't have x86_64 windows to check it, can you please run cmake configure with explicit specifying generator:

```sh
cmake -B build  -G "Unix Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141```

or

```sh
cmake -B build  -G "MSYS Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141```

pvvx, 2023-11-24:
https://github.com/devbis/tc32.git does not contain cmake.exe, nmake.exe, make.exe, ... and other files for assembly

devbis, 2023-11-24:
@pvvx don't mess up toolchain and build essentials

pvvx, 2023-11-24:
There are different "build tools and foundations" on Windows - MSYS, MinGW, Visual studio, ...

pvvx, 2023-11-24:
```
git clone https://github.com/devbis/tc32.git -b windows
```
Contains broken exe files.
Example:  https://github.com/devbis/tc32/raw/windows/tc32-elf/bin/gcc.exe

breaver, 2023-11-24:
> @breaver I don't have x86_64 windows to check it, can you please run cmake configure with explicit specifying generator:
> 
> ```shell
> cmake -B build  -G "Unix Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
> ```
> 
> or
> 
> ```shell
> cmake -B build  -G "MSYS Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
> ```

Tried both ways. Do not work:

```
D:\Downloads\z03mmc>cmake -B build  -G "Unix Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
CMake Error: Error: generator : Unix Makefiles
Does not match the generator used previously: NMake Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.

D:\Downloads\z03mmc>cmake -B build  -G "MSYS Makefiles" -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
CMake Error: Error: generator : MSYS Makefiles
Does not match the generator used previously: NMake Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
```

Sorry i am not an expert on this, what i now did instead, is using the binary file which is linked in the [release ](https://github.com/devbis/z03mmc/releases)section

I guess building the binary is not necessary then ?!

devbis, 2023-11-24:
Yes, building is not necessary, you may take a binary from the release. To build with a different generator you need to delete build folder in advance

breaver, 2023-11-24:
Then this topic can be closed. Sorry and thanks for your help!

closed 2023-11-24 by breaver

devbis, 2023-11-26:
> Contains broken exe files.

toolchain for windows is synced with files from IDE setup.

pvvx, 2023-11-26:
> toolchain for windows is synced with files from IDE setup.

But during compilation an error occurs (crash in the exe file). Telink IDE does not have nmake and many other utilities when running CMAKE scripts.

devbis, 2023-11-26:
Use either visual studio for nmake or other generators. Cmake supports many of them

pvvx, 2023-11-26:
I can assemble anything without any problems. I use dozens of different microcontrollers...
But according to your instructions, users will not be able to build the project in Windows.

devbis, 2023-11-26:
So, can you suggest any solution for cmake on windows?

pvvx, 2023-11-26:
Requires installation of MSYS or MinGW, as well as Python and git.
PATH path settings are individual. There are many MSYS MinGW packages...

In the end, it’s better to use something ready-made. For example, 'Telink IoT Studio' and throw away СMAKE - Significantly simplifies the assembly option and reduces the time for this action..