devbis/z03mmc #74: Return to Bluetooth firmware: 1141-0203-99993001-ATC_v46.zigbee not loading

3dK, 2023-11-23:
After flashing with latest firmware [z03mmc.bin](https://github.com/devbis/z03mmc/releases/download/1.0.6/z03mmc.bin), I would like to go back to bluetooth.
However, followed the guide on how to flash back to bluetooth firmware, but ZHA is not loading [1141-0203-99993001-ATC_v46.zigbee](assets/1141-0203-99993001-ATC_v46.zigbee) firmware:
`DEBUG (SyncWorker_6) [zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): /home/homeassistant/.homeassistant/zigpy_ota/1141-0203-99993001-ATC_v46.zigbee, version: 2576953345, hw_ver: (None, None), OTA string: Telink OTA BLE device
DEBUG (MainThread) [zigpy.zcl] [0x4F9B:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=4417, image_type=522, current_file_version=17903617, hardware_version=None, model='LYWSD03MMC'
DEBUG (MainThread) [zigpy.zcl] [0x4F9B:1:0x0019] No OTA image is available
`
Tried as well the [1141-0203-10063001-z03mmc.zigbee](https://github.com/devbis/z03mmc/releases/download/1.0.6/1141-0203-10063001-z03mmc.zigbee) and it wasn't uploaded.

I didn't flashed transitional custom bluetooth firmware [ATC_ota_400000](./assets/ATC_ota_40000.bin) before going to zigbee, since already had ATC_4.5.bin flashed on the device; maybe that's why?

devbis, 2023-11-23:
It looks like you flashed firmware from pvvx, not from this project. To revert you may use this binary https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v46.zigbee (0x020a)

pvvx, 2023-11-23:
https://github.com/pvvx/ZigbeeTLc#zigbee-ota

3dK, 2023-11-24:
Thank you all, firmware [0x020a](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v46.zigbee) did the trick for reverting back to ATC firmware: [LYWSD03MMC Custom Firmware Version 4.5](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v45.bin) -> [z03mmc.bin](https://github.com/devbis/z03mmc/releases/download/1.0.6/z03mmc.bin) -> [1141-020a-99993001-ATC_v46.zigbee (0x020a)](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v46.zigbee)!

3dK, 2023-11-24:
> https://github.com/pvvx/ZigbeeTLc#zigbee-ota

Is this similar to @devbis Zigbee implementation?

closed 2023-11-24 by 3dK