devbis/z03mmc #78: Enable Fahrenheit in ZHA

scarolan, 2023-11-29:
I'm not seeing a cluster for toggling between C and F.

I'm running the latest 1.0.6 firmware.  Is it possible to change the display to Fahrenheit with ZHA?

devbis, 2023-11-30:
While I'm figuring out why it is not exposed to UI, you may try ZHA toolkit https://github.com/mdeweerd/zha-toolkit#attr_write-writeread-an-attribute-value and write 1 (Fahrenheit) to 
- endpoint 1
- cluster 0x0204
- attribute 0x0000
- attr_type 0x30 
- value 1

scarolan, 2023-11-30:
Thanks @devbis - I tried the workaround but ZHA toolkit throws an error

[image]

Also am not seeing a 0x0204 cluster - there is an 0x0402 though?

[image]

devbis, 2023-12-01:
You should use `0x0204 Thermostat User Interface Configuration`, which is absent in the list.
If you're sure you're using 1.0.6, it could be a filter on ZHA side. 
Though, it is mentioned in zigpy: https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/hvac.py#L658-L670

pvvx, 2023-12-01:
https://github.com/mdeweerd/zha-toolkit/blob/main/custom_components/zha_toolkit/utils.py#L444
Log:
_Exception: No endpoint None and no cluster 0x0204 for 'a4:c1:38:0b:5e:ed:c6:97'_

scarolan, 2023-12-01:
No worries, I can wait.  The degrees on the display is not mission critical for my use case.

devbis, 2023-12-27:
@scarolan At last I was able to check switching to degrees Fahrenheit in ZHA. Unfortunately, raw zha just ignores cluster 0x0204, so I used zha-toolkit. (ieee - any entity from the device, battery, temperature, humidity, doesn't matter)

```yaml
service: zha_toolkit.attr_write
data:
  ieee: sensor.xiaomi_lywsd03mmc_battery
  cluster: 516
  attribute: 0
  attr_val: "1"
```

I used developer mode to run services. Can you please try it?

[image]

pvvx, 2023-12-27:
> Unfortunately, raw zha just ignores cluster 0x0204, so I used zha-toolkit. 

The cluster 0x0204 is not declared in the lists in the firmware.
https://github.com/pvvx/ZigbeeTLc/issues/28

scarolan, 2024-01-05:
Nice!  What should I put for "Target Cluster"?  I tried 516 like the example but didn't get a result.

scarolan, 2024-01-28:
I'm good with BLE for now, will keep my single Celsius Zigbee device for quick reference ;-)

closed 2024-01-28 by scarolan

devbis, 2024-02-07:
@scarolan The latest firmware supports switching via 1 sec pressing on reset-gnd contacts. 
And switching via cluster attribute is also fixed.

scarolan, 2024-02-15:
Nice work, @devbis!