devbis/z03mmc #82: Нет данных от датчика

sanyafifa, 2023-12-01:
Добрый! 
прошил LYWSD03MMC (20231109), в Z2M сразу добавился.
Проблема:  данные темп. и влажности не прилетают. Остальное регулярно обновляется.
[image]

[image]
[image]
[image]

несколько раз переспарил. Прошу о помощи :)

z2m ver 1.34.0 - dev

devbis, 2023-12-01:
Если в Dev Console считать данные из temperature/measuredValue, есть ли данные?

sanyafifa, 2023-12-01:
> Если в Dev Console устройства считать данные из кластера temperature/measuredValue, есть ли данные?

[image]
подскажите пож-та как это сделать?

devbis, 2023-12-01:
Выглядит, что данные есть, но бининг либо не до конца прошёл, либо какие-то проблемы с сетью. Датчик цепляется на координатор или на какой-то роутер в сети?

sanyafifa, 2023-12-01:
> Выглядит, что данные есть, но либо биндинг не до конца прошёл, либо какие-то проблемы с сетью. Датчик цепляется на координатор или на какой-то роутер в сети?

[image]

vladimir1408v, 2024-01-21:
Did you manage to solve the problem?  I have the same