devbis/z03mmc #88: TypeError: getDevVersion(...).then is not a function

sander816, 2023-12-08:
i am getting this error? Anbyone know what it means?

10:30:57: Searching for devices
10:31:06: Connecting to: ATC_F9DEA5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Reconnect 1 from 5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Reconnect 2 from 5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Reconnect 3 from 5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Reconnect 4 from 5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Reconnect 5 from 5
10:31:11: TypeError: getDevVersion(...).then is not a function
10:31:11: Something went wrong, too many reconnects

ProTofik, 2024-02-13:
Hey I have the same problem:
[image]

I tried to connect both from my Android phone and from my laptop, with exactly the same error. I have nine LYWSD03MMC and every single one gives me the same error. I am currently running ATC firmware on them.

Aslanoglu, 2024-03-04:
Same issue here...

ProTofik, 2024-04-07:
Ok I managed to solve this issue and successfully join it to my Zigbee2mqtt network. I was running ATC firmware, so I did the following:

1. Replace the battery with a brand new one
2. Go to pvvx's github, download the stock firmware binary: https://github.com/pvvx/ZigbeeTLc
3. Flash it using Telink's flasher: https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
4. Then follow this guide to flash devbis' zigbee firmware: https://smarthomescene.com/guides/convert-xiaomi-lywsd03mmc-from-bluetooth-to-zigbee/